muccc/gr-iridium #1: Airspy instead of HackRF?

dsound, 2016-05-01:
I'm awaiting the arrival of a HackRF but thought I'd try with an Airspy. Since the .conf file uses 'osmo-sdr source' I thought it might work. When I enter: 

`iridium-extractor -D 4 examples/hackrf.conf | grep "A:OK" > output.bits
`

I get:

`Traceback (most recent call last):
  File "/usr/local/bin/iridium-extractor", line 130, in <module>
    items = config.items("osmosdr-source")
  File "/usr/lib/python2.7/ConfigParser.py", line 642, in items
    raise NoSectionError(section)
ConfigParser.NoSectionError: No section: 'osmosdr-source'`

coddingtonbear, 2016-05-02:
I can't speak to whether this will work for an Airspy, but just glancing at this, it _looks_ like maybe you're not running this from the root of where you've cloned this.  The error it's raising is that it didn't find the section it's looking for in your config, which is what will happen if `examples/hackrf.conf` isn't actually the path to the configuration file.

dsound, 2016-05-02:
Thanks. I just deleted that install and tried again. I'm running as 'root' since I'm using Kali. When getting to 'make' I get this very large output:

`-- Build type not specified: defaulting to release.
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   filesystem
--   system
Checking for GNU Radio Module: RUNTIME
 * INCLUDES=/usr/include
 * LIBS=/usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so
GNURADIO_RUNTIME_FOUND = TRUE
Checking for GNU Radio Module: FFT
 * INCLUDES=/usr/include
 * LIBS=/usr/lib/x86_64-linux-gnu/libgnuradio-fft.so;/usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so
GNURADIO_FFT_FOUND = TRUE
Checking for GNU Radio Module: FILTER
 * INCLUDES=/usr/include
 * LIBS=/usr/lib/x86_64-linux-gnu/libgnuradio-filter.so;/usr/lib/x86_64-linux-gnu/libgnuradio-fft.so;/usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so
GNURADIO_FILTER_FOUND = TRUE
Checking for GNU Radio Module: VOLK
 * INCLUDES=/usr/include
 * LIBS=/usr/lib/x86_64-linux-gnu/libvolk.so
GNURADIO_VOLK_FOUND = TRUE
CMake Warning (dev) at /usr/lib/x86_64-linux-gnu/cmake/gnuradio/GrTest.cmake:45 (get_target_property):
  Policy CMP0026 is not set: Disallow use of the LOCATION target property.
  Run "cmake --help-policy CMP0026" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  The LOCATION property should not be read from target "test-iridium".  Use
  the target name directly with add_custom_command, or use the generator
  expression $<TARGET_FILE>, as appropriate.

Call Stack (most recent call first):
  lib/CMakeLists.txt:87 (GR_ADD_TEST)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- 
-- Checking for module SWIG
-- Found SWIG version 3.0.7.
CMake Warning (dev) at /usr/lib/x86_64-linux-gnu/cmake/gnuradio/GrTest.cmake:45 (get_target_property):
  Policy CMP0026 is not set: Disallow use of the LOCATION target property.
  Run "cmake --help-policy CMP0026" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  The LOCATION property should not be read from target "gnuradio-iridium".
  Use the target name directly with add_custom_command, or use the generator
  expression $<TARGET_FILE>, as appropriate.

Call Stack (most recent call first):
  python/CMakeLists.txt:47 (GR_ADD_TEST)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at /usr/lib/x86_64-linux-gnu/cmake/gnuradio/GrTest.cmake:45 (get_target_property):
  Policy CMP0045 is not set: Error on non-existent target in
  get_target_property.  Run "cmake --help-policy CMP0045" for policy details.
  Use the cmake_policy command to set the policy and suppress this warning.

  get_target_property() called with non-existent target "/usr/bin/python2".
Call Stack (most recent call first):
  python/CMakeLists.txt:47 (GR_ADD_TEST)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at /usr/lib/x86_64-linux-gnu/cmake/gnuradio/GrTest.cmake:45 (get_target_property):
  Policy CMP0045 is not set: Error on non-existent target in
  get_target_property.  Run "cmake --help-policy CMP0045" for policy details.
  Use the cmake_policy command to set the policy and suppress this warning.

  get_target_property() called with non-existent target
  "/root/gr-iridium/python/qa_fft_burst_tagger.py".
Call Stack (most recent call first):
[... 219 lines cut ...]

CMake Warning (dev) at /usr/lib/x86_64-linux-gnu/cmake/gnuradio/GrTest.cmake:45 (get_target_property):
  Policy CMP0045 is not set: Error on non-existent target in
  get_target_property.  Run "cmake --help-policy CMP0045" for policy details.
  Use the cmake_policy command to set the policy and suppress this warning.

  get_target_property() called with non-existent target "/usr/bin/python2".
Call Stack (most recent call first):
  python/CMakeLists.txt:54 (GR_ADD_TEST)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at /usr/lib/x86_64-linux-gnu/cmake/gnuradio/GrTest.cmake:45 (get_target_property):
  Policy CMP0045 is not set: Error on non-existent target in
  get_target_property.  Run "cmake --help-policy CMP0045" for policy details.
  Use the cmake_policy command to set the policy and suppress this warning.

  get_target_property() called with non-existent target
  "/root/gr-iridium/python/qa_pdu_round_robin.py".
Call Stack (most recent call first):
  python/CMakeLists.txt:54 (GR_ADD_TEST)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Configuring done
-- Generating done
`

coddingtonbear, 2016-05-02:
@dsound -- this isn't my project, but I still really recommend posting each problem you're running into as a _separate_ issue.  The first problem you posted is easily solved by making sure you're typing in the path to `hackrf.conf` properly; you don't need to re-install.  I'm not clear on what message you're trying to highlight in your second comment, but everything seems fine there.

dsound, 2016-05-02:
Sorry, I thought some of that was error but apparently not. So for the first issue, I need to run `iridium-extractor -D 4 examples/hackrf.conf | grep "A:OK" > output.bits` in the folder containing the .conf file?

coddingtonbear, 2016-05-02:
You need to make sure that `examples/hackrf.conf` is the path to the configuration file, yes.  If you are in the same folder as `hackrf.conf` you'd run `iridium-extractor -D 4 hackrf.conf | grep "A:OK" > output.bits`.  If you're somewhere else, you'd run `iridium-extractor -D 4 /whatever/the/path/to/the/configuration/file/is | grep "A:OK" > output.bits`.

dsound, 2016-05-02:
Just got my HackRF and am trying in directory with the .conf file. Have the HackRF running through a filter and into an antenna built for Iridium sitting on a pizza pan 15 stories high by the window. Does this mean it's recording Iridium data?

`gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.9
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy redpitaya 
Using HackRF One with firmware 2014.08.1 
(RF) Gain: 14.0 (Requested 20)
IF Gain: 40.0 (Requested 40)
BB Gain: 20.0 (Requested 20)
Bandwidth: 10000000.0 (Requested 10000000)
OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOO1462216233 | i:   0/s | i_avg:   0/s | q:    0 | q_max:    0 | o:   0/s | ok:   0% | ok:   0/s | ok_avg:   0% | ok:          0 | ok_avg:   0/s | d: 0
ODetector in burst squelch at 49.206612
1462216293 | i:   0/s | i_avg:   0/s | q:    0 | q_max:   12 | o:   0/s | ok:  22% | ok:   0/s | ok_avg:  22% | ok:          6 | ok_avg:   0/s | d: 0
1462216353 | i:   0/s | i_avg:   0/s | q:    0 | q_max:    1 | o:   0/s | ok:   0% | ok:   0/s | ok_avg:  20% | ok:          6 | ok_avg:   0/s | d: 0
1462216413 | i:   1/s | i_avg:   0/s | q:    0 | q_max:   64 | o:   1/s | ok:   0% | ok:   0/s | ok_avg:   5% | ok:          6 | ok_avg:   0/s | d: 0
`

schneider42, 2016-05-02:
`iridium-extractor` actually emits `ConfigParser.NoSectionError: No section: 'osmosdr-source'` if the file can not be found, so @coddingtonbear is correct.

I just fixed this in 69d69cbdeab0bc1635d4ef589932c6c3ac76322d.

Regarding the Airspy that @dsound  wants to use:
The default hackrf.conf uses a 12 MSPS sample rate. The Airspy only suports 10 MSPS with 9 MHz bandwidth.

I've just pushed some more example configurations. One which I think should work for the Airspy (I don't have one), and one which uses the HackRF with a smaller sample rate (54f125c94c297821628ac5c3bc1765cfc28a8cc3).

schneider42, 2016-05-02:
@dsound: Yes it is receiving some iridium, very poorly though.

Among other things, it runs into a burst squelch. You should check with [fosphor](http://sdr.osmocom.org/trac/wiki/fosphor), [sdrangelove](http://sdr.osmocom.org/trac/wiki/sdrangelove) or [inspectrum](https://github.com/miek/inspectrum) if your spectrum is clean.

dsound, 2016-05-02:
Ok good to know. Maybe things will improve when I get the second filter and LNA4ALL to add in series. I live in a very dense RF area (Brooklyn, NY) so filtering will be a must.

closed 2016-05-14 by schneider42