muccc/gr-iridium #2: Ability to specify RF Device

coddingtonbear, 2016-05-02:
I just happen to be using this on a machine that has both a HackRF and an RTLSDR connected, and at the moment I have to unplug the RTLSDR in order for gr-iridium to select the proper device.  Is there a simple way for me to select the proper radio device to use from the command-line?

dsound, 2016-05-23:
When I run ` iridium-extractor -D 4 airspy.conf | grep "A:OK" > output.bits` from the examples folder, I get the following error:

`gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.9
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy redpitaya 
Using AirSpy NOS v1.0.0-rc5-0-g648c14f 2015-05-20, samplerates: 2.5M 10M 
(RF) Gain: 15.0 (Requested 15)
IF Gain: 7.0 (Requested 7)
Traceback (most recent call last):
  File "/usr/local/bin/iridium-extractor", line 193, in <module>
    verbose=verbose)
  File "/usr/local/lib/python2.7/dist-packages/iridium/iridium_extractor_flowgraph.py", line 128, in __init__
    source.set_mix_gain(mix_gain, 0)
AttributeError: 'source_sptr' object has no attribute 'set_mix_gain'

Is "mix_gain" not a valid parameter name?
`

coddingtonbear, 2016-05-23:
@dsound -- I'm a little confused; is there a relationship between the issue you're having and this existing open issue?

dsound, 2016-05-23:
Well it looks like an airspy.conf file was added for airspy use. When I run it, I get an error saying no such parameter as ‘mix_gain’ - something along those lines - I’m not at that computer right now.

schneider42, 2016-05-23:
@dsound: Please provide the full output

schneider42, 2016-05-23:
Ok, I see that you've already provided some output above.

Can you tell which version of gr-osmosdr is installed on your system? I don't have an airspy, so I have to code this blind...

dsound, 2016-05-24:
Here is the output when I run 'iridium-extractor -D 4 airspy.conf | grep
"A:OK" > output.bits' in examples folder:

gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.9
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf
rfspace airspy redpitaya
Using AirSpy NOS v1.0.0-rc5-0-g648c14f 2015-05-20, samplerates: 2.5M 10M
(RF) Gain: 15.0 (Requested 15)
IF Gain: 7.0 (Requested 7)
Traceback (most recent call last):
  File "/usr/local/bin/iridium-extractor", line 193, in <module>
    verbose=verbose)
  File
"/usr/local/lib/python2.7/dist-packages/iridium/iridium_extractor_flowgraph.py",
line 128, in __init__
    source.set_mix_gain(mix_gain, 0)
AttributeError: 'source_sptr' object has no attribute 'set_mix_gain'

schneider42, 2016-08-14:
@dsound: Issue #6 deals with your isse.

@coddingtonbear: Have a look at the current hackrf config files. There is a (commented) line which you can use to tell the system to use the HackRF (and which one).

closed 2016-08-14 by schneider42

neutral01, 2019-09-24:
Hi,

Is the SDR Play RSP1A supported by GR-Iridium?
If not, would it be possible to implement it?

Thank you

Andrew.