muccc/gr-iridium #5: ImportError: No module named iridium.iridium_extractor_flowgraph

Geoff160, 2016-05-24:
Hi there,

I have compiled the source here using the included instructions, however when I go to run "iridium_extractor" at the CLI, I get this error:
```
Traceback (most recent call last):
  File "/usr/local/bin/iridium-extractor", line 10, in <module>
    import iridium.iridium_extractor_flowgraph
ImportError: No module named iridium.iridium_extractor_flowgraph
```

Any ideas on why Python can't find the modules?

Thanks,
Geoff

schneider42, 2016-05-24:
I might sound stupid, but can you try a `sudo ldconfig`?

Geoff160, 2016-05-25:
Nothing. It still can't find the iridium module.

schneider42, 2016-08-14:
The only other advice I can give is to recompile GNURadio from source and try again.

jbarg, 2016-08-27:
Have the same problem. recompiling GNURadio did not help

jbarea, 2016-09-03:
I'm getting the same error trying to get it running under mac os x. Any clue?

schneider42, 2016-09-04:
Does https://github.com/muccc/gr-iridium/issues/7#issuecomment-244175134 help in your case?

eimann, 2016-10-01:
I do export PYTHONPATH=/usr/local/lib/python2.7/site-packages:/usr/lib/python2.7/site-packages to fix this.

jbarea, 2018-01-16:
Have read in the docs and on posts over the net, but still getting the error "ImportError: No module named iridium.iridium_extractor_flowgraph" on osx. I'm using osx 10.11.6.

rkristijanto, 2020-05-26:
It seems that it is trying to import this statement here in **iridium-extractor** line 10, and it failed to find it.

`import iridium.iridium_extractor_flowgraph`

So, when I have a look where it installed the packages I notice this is the file it is trying to import

`-- Installing: /usr/local/lib/python3/dist-packages/iridium/iridium_extractor_flowgraph.py`

So I just added PYTHONPATH like below : 

`export PYTHONPATH=/usr/local/lib/python3/dist-packages`

And it fixes it ... :+1: 

Let me know if it works for you ... basically just have a look at the output closely when you run **make install**

Here is where it installed mine 

```
(base) ronald@ronald-laptop:~/Documents/Mentahan/gr-iridium-master/build$ sudo make install
[sudo] password for ronald: 
[ 38%] Built target gnuradio-iridium
[ 42%] Built target pygen_apps_69ec4
[ 47%] Built target doxygen_target
[ 57%] Built target _iridium_swig_doc_tag
[ 66%] Built target iridium_swig_swig_doc
[ 71%] Built target iridium_swig_swig_compilation
[ 80%] Built target iridium_swig
[ 90%] Built target pygen_swig_3005b
[100%] Built target pygen_python_d1885
Install the project...
-- Install configuration: "Release"
-- Installing: /usr/local/lib/cmake/iridium/iridiumConfig.cmake
-- Installing: /usr/local/include/iridium/api.h
-- Installing: /usr/local/include/iridium/fft_burst_tagger.h
-- Installing: /usr/local/include/iridium/iuchar_to_complex.h
-- Installing: /usr/local/include/iridium/tagged_burst_to_pdu.h
-- Installing: /usr/local/include/iridium/burst_downmix.h
-- Installing: /usr/local/include/iridium/pdu_null_sink.h
-- Installing: /usr/local/include/iridium/iridium_qpsk_demod_cpp.h
-- Installing: /usr/local/include/iridium/pdu_round_robin.h
-- Installing: /usr/local/lib/x86_64-linux-gnu/libgnuradio-iridium.so.v1.0-compat-xxx-xunknown
-- Installing: /usr/local/lib/x86_64-linux-gnu/libgnuradio-iridium.so.1.0.0git
-- Set runtime path of "/usr/local/lib/x86_64-linux-gnu/libgnuradio-iridium.so.v1.0-compat-xxx-xunknown" to ""
-- Installing: /usr/local/lib/x86_64-linux-gnu/libgnuradio-iridium.so
-- Installing: /usr/local/lib/cmake/iridium/gnuradio-iridiumTargets.cmake
-- Installing: /usr/local/lib/cmake/iridium/gnuradio-iridiumTargets-release.cmake
-- Installing: /usr/local/lib/cmake/iridium/gnuradio-iridiumConfig.cmake
-- Installing: /usr/local/bin/iridium-extractor
-- Up-to-date: /usr/local/share/doc/gr-iridium/xml
-- Installing: /usr/local/share/doc/gr-iridium/xml/combine.xslt
-- Installing: /usr/local/share/doc/gr-iridium/xml/namespaceiridium.xml
-- Installing: /usr/local/share/doc/gr-iridium/xml/namespacegr_1_1iridium.xml
-- Installing: /usr/local/share/doc/gr-iridium/xml/structgr_1_1iridium_1_1burst.xml
-- Installing: /usr/local/share/doc/gr-iridium/xml/classgr_1_1iridium_1_1iuchar__to__complex.xml
-- Installing: /usr/local/share/doc/gr-iridium/xml/classgr_1_1iridium_1_1pdu__null__sink__impl.xml
-- Installing: /usr/local/share/doc/gr-iridium/xml/classgr_1_1iridium_1_1iuchar__to__complex__impl.xml
-- Installing: /usr/local/share/doc/gr-iridium/xml/dir_f2cd86917185299abf5a1f0679072f3c.xml
-- Installing: /usr/local/share/doc/gr-iridium/xml/pdu__null__sink_8h.xml
-- Installing: /usr/local/share/doc/gr-iridium/xml/classgr_1_1iridium_1_1pdu__round__robin.xml
-- Installing: /usr/local/share/doc/gr-iridium/xml/classgr_1_1iridium_1_1iridium__qpsk__demod__cpp.xml
-- Installing: /usr/local/share/doc/gr-iridium/xml/group__block.xml
-- Installing: /usr/local/share/doc/gr-iridium/xml/index.xsd
-- Installing: /usr/local/share/doc/gr-iridium/xml/namespacegr.xml
-- Installing: /usr/local/share/doc/gr-iridium/xml/dir_d44c64559bbebec7f509842c48db8b23.xml
-- Installing: /usr/local/share/doc/gr-iridium/xml/iuchar__to__complex_8h.xml
-- Installing: /usr/local/share/doc/gr-iridium/xml/tagged__burst__to__pdu_8h.xml
-- Installing: /usr/local/share/doc/gr-iridium/xml/group__defs_8dox.xml
-- Installing: /usr/local/share/doc/gr-iridium/xml/main__page_8dox.xml
-- Installing: /usr/local/share/doc/gr-iridium/xml/pdu__round__robin_8h.xml
[... 325 lines cut ...]
-- Installing: /usr/local/lib/python3/dist-packages/iridium/__init__.pyc
-- Installing: /usr/local/lib/python3/dist-packages/iridium/iridium_extractor_flowgraph.pyc
-- Installing: /usr/local/lib/python3/dist-packages/iridium/iridium_qpsk_demod.pyc
-- Installing: /usr/local/lib/python3/dist-packages/iridium/gr_iridium.pyc
-- Installing: /usr/local/lib/python3/dist-packages/iridium/gr_iridium_demod.pyc
-- Installing: /usr/local/lib/python3/dist-packages/iridium/iridium_frame_printer.pyc
-- Installing: /usr/local/lib/python3/dist-packages/iridium/frame_sorter.pyc
-- Installing: /usr/local/lib/python3/dist-packages/iridium/__init__.pyo
-- Installing: /usr/local/lib/python3/dist-packages/iridium/iridium_extractor_flowgraph.pyo
-- Installing: /usr/local/lib/python3/dist-packages/iridium/iridium_qpsk_demod.pyo
-- Installing: /usr/local/lib/python3/dist-packages/iridium/gr_iridium.pyo
-- Installing: /usr/local/lib/python3/dist-packages/iridium/gr_iridium_demod.pyo
-- Installing: /usr/local/lib/python3/dist-packages/iridium/iridium_frame_printer.pyo
-- Installing: /usr/local/lib/python3/dist-packages/iridium/frame_sorter.pyo
-- Installing: /usr/local/share/gnuradio/grc/blocks/iridium_fft_burst_tagger.block.yml
-- Installing: /usr/local/share/gnuradio/grc/blocks/iridium_iuchar_to_complex.block.yml
-- Installing: /usr/local/share/gnuradio/grc/blocks/iridium_tagged_burst_to_pdu.block.yml
[... cut ...]
-- Installing: /usr/local/share/gnuradio/grc/blocks/iridium_iridium-qpsk-demod.block.yml
-- Installing: /usr/local/share/gnuradio/grc/blocks/iridium_pdu_null_sink.block.yml
-- Installing: /usr/local/share/gnuradio/grc/blocks/iridium_iridium_qpsk_demod_cpp.block.yml
-- Installing: /usr/local/share/gnuradio/grc/blocks/iridium_pdu_round_robin.block.yml
-- Installing: /usr/local/share/gnuradio/grc/blocks/iridium_iridium_frame_printer.block.yml
-- Installing: /usr/local/share/gnuradio/grc/blocks/iridium_frame_sorter.block.yml
```

Omid137, 2020-12-07:
> ```
> (base) ronald@ronald-laptop:~$ iridium-extractor 
> Traceback (most recent call last):
>   File "/usr/local/bin/iridium-extractor", line 10, in <module>
>     import iridium.iridium_extractor_flowgraph
> ModuleNotFoundError: No module named 'iridium'
> ```
> 
> It seems that it is trying to import this statement here in **iridium-extractor** line 10, and it failed to find it.
> 
> `import iridium.iridium_extractor_flowgraph`
> 
> So, when I have a look where it installed the packages I notice this is the file it is trying to import
> 
> `-- Installing: /usr/local/lib/python3/dist-packages/iridium/iridium_extractor_flowgraph.py`
> 
> So I just added PYTHONPATH like below :
> 
> `export PYTHONPATH=/usr/local/lib/python3/dist-packages`
> 
> And it fixes it ... +1
> 
> Let me know if it works for you ... basically just have a look at the output closely when you run **make install**
> 
> Here is where it installed mine
> 
> ```
> (base) ronald@ronald-laptop:~/Documents/Mentahan/gr-iridium-master/build$ sudo make install
> [sudo] password for ronald: 
> [ 38%] Built target gnuradio-iridium
> [ 42%] Built target pygen_apps_69ec4
> [ 47%] Built target doxygen_target
> [ 57%] Built target _iridium_swig_doc_tag
> [ 66%] Built target iridium_swig_swig_doc
> [ 71%] Built target iridium_swig_swig_compilation
> [ 80%] Built target iridium_swig
> [ 90%] Built target pygen_swig_3005b
> [100%] Built target pygen_python_d1885
> Install the project...
> -- Install configuration: "Release"
> -- Installing: /usr/local/lib/cmake/iridium/iridiumConfig.cmake
> -- Installing: /usr/local/include/iridium/api.h
> -- Installing: /usr/local/include/iridium/fft_burst_tagger.h
> -- Installing: /usr/local/include/iridium/iuchar_to_complex.h
> -- Installing: /usr/local/include/iridium/tagged_burst_to_pdu.h
> -- Installing: /usr/local/include/iridium/burst_downmix.h
> -- Installing: /usr/local/include/iridium/pdu_null_sink.h
> -- Installing: /usr/local/include/iridium/iridium_qpsk_demod_cpp.h
> -- Installing: /usr/local/include/iridium/pdu_round_robin.h
> -- Installing: /usr/local/lib/x86_64-linux-gnu/libgnuradio-iridium.so.v1.0-compat-xxx-xunknown
> -- Installing: /usr/local/lib/x86_64-linux-gnu/libgnuradio-iridium.so.1.0.0git
> -- Set runtime path of "/usr/local/lib/x86_64-linux-gnu/libgnuradio-iridium.so.v1.0-compat-xxx-xunknown" to ""
> -- Installing: /usr/local/lib/x86_64-linux-gnu/libgnuradio-iridium.so
> -- Installing: /usr/local/lib/cmake/iridium/gnuradio-iridiumTargets.cmake
> -- Installing: /usr/local/lib/cmake/iridium/gnuradio-iridiumTargets-release.cmake
> -- Installing: /usr/local/lib/cmake/iridium/gnuradio-iridiumConfig.cmake
> -- Installing: /usr/local/bin/iridium-extractor
> -- Up-to-date: /usr/local/share/doc/gr-iridium/xml
> -- Installing: /usr/local/share/doc/gr-iridium/xml/combine.xslt
> -- Installing: /usr/local/share/doc/gr-iridium/xml/namespaceiridium.xml
> -- Installing: /usr/local/share/doc/gr-iridium/xml/namespacegr_1_1iridium.xml
> -- Installing: /usr/local/share/doc/gr-iridium/xml/structgr_1_1iridium_1_1burst.xml
> -- Installing: /usr/local/share/doc/gr-iridium/xml/classgr_1_1iridium_1_1iuchar__to__complex.xml
> -- Installing: /usr/local/share/doc/gr-iridium/xml/classgr_1_1iridium_1_1pdu__null__sink__impl.xml
> -- Installing: /usr/local/share/doc/gr-iridium/xml/classgr_1_1iridium_1_1iuchar__to__complex__impl.xml
> -- Installing: /usr/local/share/doc/gr-iridium/xml/dir_f2cd86917185299abf5a1f0679072f3c.xml
> -- Installing: /usr/local/share/doc/gr-iridium/xml/pdu__null__sink_8h.xml
> -- Installing: /usr/local/share/doc/gr-iridium/xml/classgr_1_1iridium_1_1pdu__round__robin.xml
> -- Installing: /usr/local/share/doc/gr-iridium/xml/classgr_1_1iridium_1_1iridium__qpsk__demod__cpp.xml
> -- Installing: /usr/local/share/doc/gr-iridium/xml/group__block.xml
[... 335 lines cut ...]
> -- Installing: /usr/local/lib/python3/dist-packages/iridium/iridium_qpsk_demod.pyc
> -- Installing: /usr/local/lib/python3/dist-packages/iridium/gr_iridium.pyc
> -- Installing: /usr/local/lib/python3/dist-packages/iridium/gr_iridium_demod.pyc
> -- Installing: /usr/local/lib/python3/dist-packages/iridium/iridium_frame_printer.pyc
> -- Installing: /usr/local/lib/python3/dist-packages/iridium/frame_sorter.pyc
> -- Installing: /usr/local/lib/python3/dist-packages/iridium/__init__.pyo
> -- Installing: /usr/local/lib/python3/dist-packages/iridium/iridium_extractor_flowgraph.pyo
> -- Installing: /usr/local/lib/python3/dist-packages/iridium/iridium_qpsk_demod.pyo
> -- Installing: /usr/local/lib/python3/dist-packages/iridium/gr_iridium.pyo
> -- Installing: /usr/local/lib/python3/dist-packages/iridium/gr_iridium_demod.pyo
> -- Installing: /usr/local/lib/python3/dist-packages/iridium/iridium_frame_printer.pyo
> -- Installing: /usr/local/lib/python3/dist-packages/iridium/frame_sorter.pyo
> -- Installing: /usr/local/share/gnuradio/grc/blocks/iridium_fft_burst_tagger.block.yml
> -- Installing: /usr/local/share/gnuradio/grc/blocks/iridium_iuchar_to_complex.block.yml
> -- Installing: /usr/local/share/gnuradio/grc/blocks/iridium_tagged_burst_to_pdu.block.yml
> -- Installing: /usr/local/share/gnuradio/grc/blocks/iridium_burst_downmix.block.yml
> -- Installing: /usr/local/share/gnuradio/grc/blocks/iridium_iridium-qpsk-demod.block.yml
> -- Installing: /usr/local/share/gnuradio/grc/blocks/iridium_pdu_null_sink.block.yml
> -- Installing: /usr/local/share/gnuradio/grc/blocks/iridium_iridium_qpsk_demod_cpp.block.yml
> -- Installing: /usr/local/share/gnuradio/grc/blocks/iridium_pdu_round_robin.block.yml
> -- Installing: /usr/local/share/gnuradio/grc/blocks/iridium_iridium_frame_printer.block.yml
> -- Installing: /usr/local/share/gnuradio/grc/blocks/iridium_frame_sorter.block.yml
> ```

Thank you for your detailed answer. where should We add the export command? after sudo make install?

rkristijanto, 2020-12-15:
Yes, either way will work.

schneider42, 2021-10-01:
This seems to be an issue with GNURadio and how it handles the Python package installation location: https://github.com/gnuradio/gnuradio/issues/5121

schneider42, 2022-02-01:
Should be fixed in recent gr-iridium and GNURadio versions.

closed 2022-02-01 by schneider42