muccc/gr-iridium #6: AttributeError: 'source_sptr' object has no attribute 'set_mix_gain'

rhaamo, 2016-08-14:
Hello,

Trying with an AIRSPY MINI and getting:
```
iridium-extractor -D 4 airspy.conf|grep "A:OK" > output.airspy.bits
gr-osmosdr v0.1.4-72-g164a09fc (0.1.5git) gnuradio 3.7.10
built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy redpitaya 
Using AirSpy MINI v1.0.0-rc9-0-ga56adfd 2016-06-12, samplerates: 3M 6M 
(RF) Gain: 15.0 (Requested 15)
IF Gain: 7.0 (Requested 7)
Traceback (most recent call last):
  File "/usr/local/bin/iridium-extractor", line 199, in <module>
    verbose=verbose)
  File "/usr/local/lib/python2.7/dist-packages/iridium/iridium_extractor_flowgraph.py", line 129, in __init__
    source.set_mix_gain(mix_gain, 0)
AttributeError: 'source_sptr' object has no attribute 'set_mix_gain'
```

Any ideas what could be wrong ?

schneider42, 2016-08-14:
The airspy should support this gain, I'm not sure why you are running into this exception.

I've just pushed a version which will first check if a gain is available before setting it. It should print a warning in your case. It will also print the available gains in this case.

Can you try the current master and see if it solves your problem?

rhaamo, 2016-08-14:
Thanks, I have just pulled and make/make install and it seems to be okay now :
```
gr-osmosdr v0.1.4-72-g164a09fc (0.1.5git) gnuradio 3.7.10
built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy redpitaya 
Using AirSpy MINI v1.0.0-rc9-0-ga56adfd 2016-06-12, samplerates: 3M 6M 
(RF) Gain: 15.0 (Requested 15)
MIX Gain: 7.0 (Requested 7)
IF Gain: 7.0 (Requested 7)
Bandwidth: 10000000.0 (Requested 9000000)
1471203290 | i:   0/s | i_avg:   0/s | q:    0 | q_max:    0 | o:   0/s | ok:   0% | ok:   0/s | ok_avg:   0% | ok:          0 | ok_avg:   0/s | d: 0
1471203291 | i:   0/s | i_avg:   0/s | q:    0 | q_max:    0 | o:   0/s | ok:   0% | ok:   0/s | ok_avg:   0% | ok:          0 | ok_avg:   0/s | d: 0
1471203292 | i:   0/s | i_avg:   0/s | q:    0 | q_max:    0 | o:   0/s | ok:   0% | ok:   0/s | ok_avg:   0% | ok:          0 | ok_avg:   0/s | d: 0
```

closed 2016-09-10 by schneider42