muccc/gr-iridium #7: 'module' object has no attribute 'fft_burst_tagger'

throwaway234, 2016-08-31:
Environment: Clean fedora 23 x86_64 install, GnuRadio built with build-gnuradio script.

After building gr-iridium and installing, iridium-extractor fails because the iridium module does not contain 'fft_burst_tagger':

`
[user@iridium gr-iridium]$ iridium-extractor  -D 4 examples/rtl-sdr.conf  linux; GNU C++ version 5.3.1 20160406 (Red Hat 5.3.1-6); Boost_105800; UHD_003.011.000.git-47-g3e352429

gr-osmosdr v0.1.4-75-gae686c46 (0.1.5git) gnuradio 3.7.10.1
built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy redpitaya 

FATAL: No supported devices found to pick from.

Trying to fill up 1 missing channel(s) with null source(s).
This is being done to prevent the application from crashing
due to gnuradio bug #528.

(RF) Gain: 0.0 (Requested 19)
WARNING: Gain IF not supported by source!
Supported gains: ()
Bandwidth: 0.0 (Requested 0)
Traceback (most recent call last):
  File "/usr/local/bin/iridium-extractor", line 199, in <module>
    verbose=verbose)
  File "/usr/local/lib64/python2.7/site-packages/iridium/iridium_extractor_flowgraph.py", line 173, in __init__
    self._fft_burst_tagger = iridium.fft_burst_tagger(center_frequency=self._center_frequency,
AttributeError: 'module' object has no attribute 'fft_burst_tagger'
`

It seems the built iridium module does not seem to have any of the SWIG stuff, only the pure python code:
`
[user@iridium gr-iridium]$ python
Python 2.7.11 (default, Aug  9 2016, 15:45:42) 
[GCC 5.3.1 20160406 (Red Hat 5.3.1-6)] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import iridium
>>> dir(iridium)
['__builtins__', '__doc__', '__file__', '__name__', '__package__', '__path__', 'gr_iridium', 'gr_iridium_demod', 'iridium_qpsk_demod']
`

Build log (no obviously relevant errors that I can see):
[buildfailure.txt](https://github.com/muccc/gr-iridium/files/448468/buildfailure.txt)

schneider42, 2016-09-01:
It looks like something is going very wrong with SWIG.

Can you post your `iridium_swig.py` contents?

throwaway234, 2016-09-01:
No problem:
[iridium_swig.py.txt](https://github.com/muccc/gr-iridium/files/449079/iridium_swig.py.txt)

throwaway234, 2016-09-01:
Nevermind, my bad. Fixed by adding /usr/local/lib64 to LD_LIBRARY_PATH. Closing this case, thanks!

shyamjos, 2016-09-10:
i am also having same error :( (fedora 24)

Traceback (most recent call last):
  File "/usr/local/bin/iridium-extractor", line 10, in <module>
    import iridium.iridium_extractor_flowgraph
ImportError: No module named iridium.iridium_extractor_flowgraph

also tried:  export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/local/lib6

still no luck

shyamjos, 2016-09-13:
I have fixed it by copying /usr/local/lib64/python2.7/site-packages/iridium to /usr/lib64/python2.7/site-packages/  in fedora 24

xluccianox, 2016-09-18:
I had the same issue than @throwaway234, I'm using Kali Linux 2016.1-amd64 VM.

> ed@kali:~/iridium-toolkit-master/gr-iridium-master/examples# iridium-extractor -D 4 rtl-sdr.conf 
> linux; GNU C++ version 6.2.0 20160822; Boost_106100; UHD_003.009.005-0-unknown
> 
> gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.10
> built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy soapy redpitaya 
> Detached kernel driver
> Found Rafael Micro R820T tuner
> Reattached kernel driver
> Using device #0 Realtek RTL2838UHIDIR SN: 00000001
> Detached kernel driver
> Found Rafael Micro R820T tuner
> [R82XX] PLL not locked!
> Exact sample rate is: 2000000.052982 Hz
> [R82XX] PLL not locked!
> (RF) Gain: 19.7 (Requested 19)
> WARNING: Gain IF not supported by source!
> Supported gains: ('LNA',)
> Bandwidth: 0.0 (Requested 0)
> Traceback (most recent call last):
>   File "/usr/local/bin/iridium-extractor", line 199, in <module>
>     verbose=verbose)
>   File "/usr/local/lib/python2.7/dist-packages/iridium/iridium_extractor_flowgraph.py", line 173, in __init__
>     self._fft_burst_tagger = iridium.fft_burst_tagger(center_frequency=self._center_frequency,
> **AttributeError: 'module' object has no attribute 'fft_burst_tagger'**
> Reattached kernel driver
>

I tried the export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/local/lib64 and copying /usr/local/lib64/python2.7/site-packages/iridium to /usr/lib64/python2.7/site-packages/ (suggested by @shyamjos ,not sure if it makes any sense in Kali). Both without success.

Any ideas @schneider42 ? Later today I'll try building a clean Fedora VM to discard anything else.

Thanks.

daniel4x, 2016-09-24:
Got the same issue :

```
Traceback (most recent call last):
  File "/usr/local/bin/iridium-extractor", line 199, in <module>
    verbose=verbose)
  File "/usr/local/lib/python2.7/dist-packages/iridium/iridium_extractor_flowgraph.py", line 173, in __init__
    self._fft_burst_tagger = iridium.fft_burst_tagger(center_frequency=self._center_frequency,
AttributeError: 'module' object has no attribute 'fft_burst_tagger'
```

Ubuntu 16.04 LTS .. 
I checked gnuradio and it has component named fft_burst_tagger.

Any idea?

yoelto, 2016-10-15:
Hey got exactly the same issue -
tried to add relevant directories to the path,
in the python2.7/dist-packages/iridium directory there is no fft_burst_tagger indeed but in the gr_iridium/python directory there is some qa_fft_burst_tagger or something like this.
GRC indeed has the fft_burst_tagger block.
the swig file includes fft_burst_tagger but I don't see any calling for it anywhere.
I am bit of a newbie to python as well as to gnuradio PLEASE HELP! :)

barf, 2017-12-28:
FYI resolved this by running 'ldconfig' after make install on Ubuntu 16.04 here

closed 2021-01-14 by Sec42

Unleashed1, 2025-04-17:
Hi guys if u having the same problem i solved it using this script to build gnuradio https://github.com/gnuradio/gnuradio/issues/3769
then builded gr-iridium