muccc/gr-iridium #12: Can't build in a RaspberryPI 3 - link error

alexbogus, 2016-09-11:
I' m trying to compile gr-iridium in a RaspberryPI 3 but when I try to compile it I have a link error (http://pastebin.com/Vjn7t9Ja).

Any suggestion to solve the issue?
Thanks in advance

schneider42, 2016-09-11:
Hm, looks like an incompatibility with your installed GNURadio version. How did you install GNURadio and which version is it?

alexbogus, 2016-09-17:
The GNURadio installation was via respository using apt-get. I'll try to reinstall the GNURadio and tell you, thanks for  your assistance. Regards

schneider42, 2016-11-26:
You will need at least GNU Radio 3.7.6

I've been playing with a Raspberry Pi 2 and noticed that it is not build with ARM NEON support, which will make it rather slow. It looks like the assembler does not allow to assemble all NEON instructions either, so the build of a recent version of GNU Radio also fails...

schneider42, 2016-12-03:
Please try to use an Ubuntu image: https://wiki.ubuntu.com/ARM/RaspberryPi#Ubuntu_16.04_LTS

This worked for me:
```
sudo apt-get update
sudo apt-get dist-upgrade
```
reboot
```
sudo apt-get install gnuradio-dev swig gr-osmosdr cmake
sudo git clone https://github.com/muccc/gr-iridium
cd gr-iridium
mkdir build
cd build
cmake ..
make
sudo make install
```

You can expect the Raspberry Pi 2 to be able to handle about 2 Msps.

closed 2016-12-03 by schneider42