muccc/gr-iridium #19: Overflowing some part of the chain with 6MSPS config

n0p, 2017-01-07:
I am using an airspy mini with this config:
`[osmosdr-source]
device_args='airspy=0,bias=1'
sample_rate=6000000
center_freq=1622000000
#bandwidth=6000000

# Linearity Gain
gain=18
`

And a couple of seconds after running iridium-extractor, the output starts spitting OOOOOOOOOOOOOOOOOOOOOOOOOOOOOs:
`iridium-extractor -D 4 airspy.conf | grep "A:OK" > nfs/output.bits
gr-osmosdr v0.1.4-75-gae686c46 (0.1.5git) gnuradio v3.7.10.1-190-g3f14e437
built-in source types: file osmosdr fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy redpitaya 
Using AirSpy MINI v1.0.0-rc10-0-g946184a 2016-09-19, samplerates: 3M 6M 
(RF) Gain: 18.0 (Requested 18)
Warning: Setting bandwidth to 10000000.0
1483815950 | i:   0/s | i_avg:   0/s | q:    0 | q_max:    0 | o:   0/s | ok:   0% | ok:   0/s | ok_avg:   0% | ok:          0 | ok_avg:   0/s | d: 0
1483815951 | i:   0/s | i_avg:   0/s | q:    0 | q_max:    0 | o:   0/s | ok:   0% | ok:   0/s | ok_avg:   0% | ok:          0 | ok_avg:   0/s | d: 0
OOOOOOOOOOOOOOOOOOOOOOOOOOOOOO1483815952 | i:   0/s | i_avg:   0/s | q:    0 | q_max:    1 | o:   0/s | ok:   0% | ok:   0/s | ok_avg:   0% | ok:          0 | ok_avg:   0/s | d: 0
OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOO1483815953 | i:   1/s | i_avg:   0/s | q:    0 | q_max:    1 | o:   1/s | ok:   0% | ok:   0/s | ok_avg:   0% | ok:          0 | ok_avg:   0/s | d: 0
OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOO1483815954 | i:   9/s | i_avg:   3/s | q:    0 | q_max:    4 | o:   8/s | ok:  20% | ok:   1/s | ok_avg:  15% | ok:          2 | ok_avg:   0/s | d: 0
OOOOOOOOOOOO^C`

It seems to work, but I don't know how to fix this.

schneider42, 2017-01-10:
This is strange. You are getting lots of overflows but almost no bursts detected. Is your machine fast enough to run the FFT? What does `top` say (press `H` to see which thread is consuming most of the resources)? Is the CPU busy at all?

n0p, 2017-01-10:
That may be it, I'm using a pine64 board, it has a 1.2 Ghz Quad-Core ARM Cortex A53 with 2GB of RAM and Debian 64 (armhf)

But the top shows a top usr load of 75% in any core (20 to 40% idle) and iridium-extractor using a 250% CPU, maybe I'm reading these wrongly.

n0p, 2017-01-10:
Ok, it seems that the SOC is reaching 73ºC and throttling the CPU :/

closed 2017-01-10 by n0p