muccc/gr-iridium #26: d_max_bursts calculation invalid

dillona, 2017-10-27:
I was constantly getting errors about "Detector in burst squelch".

After looking through the code a bit, I think the logic for calculating d_max_bursts in fft_burst_tagger_impl is incorrect. 
```
// Consider the signal to be invalid if more than 80%
// of all channels are in use.
d_max_bursts = (sample_rate / burst_width) * 0.8;
```

d_max_bursts is intended to be a count measured in FFT bins. However burst_width is measured in samples. As such, d_max_bursts actually winds up being measured in samples^2 per second instead of FFT bins.

I think the fix is to define d_max_bursts relative to d_burst_width (which is measured in FFT bins).

Sec42, 2017-10-29:
I'm not 100% sure about the calculation (it's @schneider42 code). Have you checked signal in these cases and confirmed it is not actually interference, but valid data that gets thrown away? If so, could you provide a short sample of this signal?

schneider42, 2017-10-30:
`d_max_bursts` gets checked against `d_bursts.size()` which is the number of currently active bursts. Both are numbers of bursts, not FFT bins. If the fft_burst_tagger reports an overflow, it means that more than 80% of  the band seems to have an active burst.

@dillona If you can share a recording which triggers this, we can have a look.

closed 2022-02-01 by schneider42