muccc/gr-iridium #28: *** Error in `/usr/bin/python2': corrupted double-linked list: 0x00cd9fe8 ***

Dizzy-Pi, 2018-04-19:
Hi, I'm getting this error now on both my linux mint 18 build and my RPi 3 Raspbian build. The full text is:  pi@raspberrypi:~/gr-iridium $ iridium-extractor -D 4 examples/rtl-sdr.conf | grep "A:OK" > output.bits
gr-osmosdr v0.1.4-98-gc653754d (0.1.5git) gnuradio 3.7.10
built-in source types: file fcd rtl rtl_tcp uhd hackrf rfspace redpitaya 
*** Error in `/usr/bin/python2': corrupted double-linked list: 0x00cd9fe8 ***

Dizzy-Pi, 2018-05-10:
Is anyone monitoring this page? I have reinstalled gr-iridium on both machines and the error persists. I get no python errors in any other programs.

Sec42, 2018-05-10:
Yes, but I can't reproduce the problem. What version of gnuradio are you using? Did you install/compile it yourself or is it a package?

Dizzy-Pi, 2018-05-10:
Hi and thanks. The issue on the Rpi has GNU Radio companion 3.7.10. I can't recall how it was installed or if it was already part of raspbian.

Sec42, 2018-05-10:
Can you try building gnuradio from (current) source and then recompiling gr-iridium?

Dizzy-Pi, 2018-05-10:
ok will do. no time left tonight but will post again tomorrow, thank you.

Dizzy-Pi, 2018-05-11:
Built gnuradio from from current source and recompiled gr-iridium. Now have "importerror: no module named osmosdr_swig". I have swig and osmosdr installled. so thats where i'm up to.

Dizzy-Pi, 2018-05-12:
I'm going to close this as it is not a bug, more of "trying to juggle and understand everything" issue. I will keep at it until it runs again. Thanks Sec42, i appreciate your work on this project.