muccc/gr-iridium #34: Error in installation: GnuradioConfig.cmake missing

pelixm, 2019-01-23:
Hello
I am trying to install gr-iridium-master on my computer running Ubuntu 14.04.3.
In order to fix prerequisites I already installed (with the Linux package installer) Gnuradio 3.7.10 which works fine, having tested it with an SDR application.
Nevertheless when running the "cmake .." as in installation instructions I get the following error messages :

*****************************************************************************
CMake Error at CMakeLists.txt:116 (find_package):
  By not providing "FindGnuradio.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Gnuradio",
  but CMake did not find one.

  Could not find a package configuration file provided by "Gnuradio"
  (requested version 3.7.6) with any of the following names:

    GnuradioConfig.cmake
    gnuradio-config.cmake

  Add the installation prefix of "Gnuradio" to CMAKE_PREFIX_PATH or set
  "Gnuradio_DIR" to a directory containing one of the above files.  If
  "Gnuradio" provides a separate development package or SDK, be sure it has
  been installed.

****************************************************************************************************

Actually the mentioned cmake files do not exist on my computer so I cannot manually insert the directory: any suggestion for workarounds?
Thanks. Mauro

neil969, 2020-04-08:
I am seeing the same issue.. I have GNUradio installed from the repository (on LinuxMint) too.
Any clues please ??? 
I can't find these files either :(

schneider42, 2020-04-09:
Did you build from the maint-v3.7 branch?
`git checkout maint-3.7`

neil969, 2020-04-09:
Thanks for the reply!! I am grateful !  
yes.. Although I have  started again and git cloned the  repository , ( and grabbed the zip) but now when I type git checkout maint-3.7  I am getting  "fatal: not a git repository (or any of the parent directories): .git error " : )
I am not any kind of expert in building, - just following instructions LOL!  I think I am missing something here though

neil969, 2020-04-10:
Just a bit further to this..
I made a VM with a clean copy of Ubuntu, and it seems you NEED to do a local compile of GnuRadio before you can build gr-iridium, otherwise you get the above missing files. 
The problem is that doing a successful build of GnuRadio is like pulling teeth, as it always seems to have *something* missing .. 
I am still failing with this :( 
Is it possible to change it so it will compile when you have a working copy of GnuRadio from the repositories?

schneider42, 2022-02-01:
Should just work with GNURadio from Ubuntu 20.04 and the maint-3.8 branch

closed 2022-02-01 by schneider42