muccc/gr-iridium #37: Activation of the Bias tee in gr-iridium

yasnieltc, 2019-09-11:
Hi, I have an rtl-sdr.com v3 dongle and it works perfectly in Windows 10 when I activate the polarization tee. But in Kali Linux 2019 it is not like that, every time I run the command ./rtl_biast the biast is enabled without problems and I can check the output of the sma connector of the 4.5 volt dongle, but when I run gr-iridium it deactivates the dongle voltage and Of course, the GPS patch antenna does not work. I hope you can help me, because I am very interested in decoding the signals of the Iridium satellite system.

codigodelbracico, 2019-09-24:
i have the same issue with hackrf

jnse, 2020-03-04:
did you try just passing bias=1 in the device string?
like for example, in your conf:
```
device_args='rtl=0,bias=1'
```
that's worked for me just fine using an rtl-sdr dongle.

codigodelbracico, 2020-04-12:
No, it doesn´t work with hackrf, i´m stating to think that my Hackrf doesn´t works