muccc/gr-iridium #38: Conf file for SDRduo

codigodelbracico, 2019-09-24:
Good afternoon everyone! I was thinking that within the most common and affordable SDR is the sdrplay duo, I wanted to know if someone has configured a conf file for that device, really, that would be a great improvement.

Greetings!

neutral01, 2019-09-25:
Hi All,

I have got an SDR Play RSP1A, I wondered if a configuration file could be made available to enable it to be compatible with GR-Iridium.

Thank you

Andrew.

codigodelbracico, 2020-04-12:
IT would be fine, developers?

Geezerjem, 2020-08-10:
Could a new .conf for SDRPlay devices be generated to supply with the main source code please ? Or at least the information for a punter to generate one. They are popular devices ( even I have an RSP1A ! :-) ), so I think that this would add to the install base. I have searched the interweb and there's nothing solid as to generating one.
Cheers,
Jem

Sec42, 2020-11-15:
Sorry, I know nothing about the SDR duo or SDR Play devices.

If the gnuradio osmosdr source has support for it, it should "just work" - just use e.g. the [airspy.conf](https://github.com/muccc/gr-iridium/blob/master/examples/airspy.conf) as starting point

If osmosdr does not support your SDR, I suggest making a recording to a raw file with your SDR-specific tool, then running that through iridium-extractor as specified in [in the readme](https://github.com/muccc/gr-iridium#extracting-iridium-frames-from-raw-data)