muccc/gr-iridium #46: shared_ptr reference don't track against gnuradio master (3.9 prep)

yakkonaut, 2020-07-01:
Not 100% sure this should go in to master, but also not sure where to put it. If possible can gr-iridium change branching to match gnuradio? add a maint-3.8/3.7 so master can track against gnuradio master?

Pull request 45 modifies the references, but may need to be rejected in favor of a repo refactor

https://github.com/muccc/gr-iridium/pull/45

schneider42, 2020-07-14:
By now I'm wondering how people are able to install any OOT module anymore given the mess with all these breaking changes between 3.7/3.8/3.9 :/

Is there a reason why you want to use gnuradio master? I consider 3.8 the stable version for "production" use and master more something for people who actually want to develop / test. What do you think about keeping gr-iridium master on 3.8 and introducing a main-3.9 branch, which eventually become master?

schneider42, 2020-07-16:
Btw: we do have main-3.8 and maint-3.7 branches.

schneider42, 2021-10-01:
This has been fixed. Thanks.

closed 2021-10-01 by schneider42