muccc/gr-iridium #47: gnuradio 3.9 breaks burst_downmix_impl reference to fft_complex

yakkonaut, 2020-07-01:
Error below, gnuradio commit https://github.com/gnuradio/gnuradio/commit/eae138c160a6fb2834eb5eda392d0ed25fbc48d0 added the `fft_complex(const fft_complex&) = delete;` line that appears to be breaking `burst_downmix_impl`. I'm trying to figure out how to fix this the right way, but it's not jumping out at me. Raising this issue to try and get some extra eyes on it.
```
Building CXX object lib/CMakeFiles/gnuradio-iridium.dir/burst_downmix_impl.cc.o
/home/user/gr-iridium/lib/burst_downmix_impl.cc: In constructor ‘gr::iridium::burst_downmix_impl::burst_downmix_impl(int, int, size_t, const std::vector<float>&, const std::vector<float>&, bool)’:
/home/user/gr-iridium/lib/burst_downmix_impl.cc:115:98: error: use of deleted function ‘gr::fft::fft_complex::fft_complex(const gr::fft::fft_complex&)’
  115 |               d_cfo_est_fft(fft::fft_complex(d_cfo_est_fft_size * d_fft_over_size_facor, true, 1))
      |                                                                                                  ^
In file included from /home/user/gr-iridium/lib/burst_downmix_impl.cc:28:
/opt/gnuradio/gnuradio/include/gnuradio/fft/fft.h:68:5: note: declared here
   68 |     fft_complex(const fft_complex&) = delete;
      |     ^~~~~~~~~~~
/home/user/gr-iridium/lib/burst_downmix_impl.cc: In member function ‘void gr::iridium::burst_downmix_impl::initialize_correlation_filter()’:
/home/user/gr-iridium/lib/burst_downmix_impl.cc:278:69: error: use of deleted function ‘gr::fft::fft_complex::fft_complex(const gr::fft::fft_complex&)’
  278 |       fft::fft_complex fft_engine = fft::fft_complex(d_corr_fft_size);
      |                                                                     ^
In file included from /home/user/gr-iridium/lib/burst_downmix_impl.cc:28:
/opt/gnuradio/gnuradio/include/gnuradio/fft/fft.h:68:5: note: declared here
   68 |     fft_complex(const fft_complex&) = delete;
      |     ^~~~~~~~~~~
/home/user/gr-iridium/lib/burst_downmix_impl.cc: In member function ‘virtual int gr::iridium::burst_downmix_impl::work(int, gr_vector_const_void_star&, gr_vector_void_star&)’:
/home/user/gr-iridium/lib/burst_downmix_impl.cc:741:5: warning: no return statement in function returning non-void [-Wreturn-type]
  741 |     }
      |     ^
make[2]: *** [lib/CMakeFiles/gnuradio-iridium.dir/build.make:122: lib/CMakeFiles/gnuradio-iridium.dir/burst_downmix_impl.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:292: lib/CMakeFiles/gnuradio-iridium.dir/all] Error 2
make: *** [Makefile:161: all] Error 2
```

yakkonaut, 2020-07-01:
It looks like there's a dangling pointer issue with that kind of instantiation of `fft::fft_complex` per comment here:
https://github.com/gnuradio/gnuradio/commit/eae138c160a6fb2834eb5eda392d0ed25fbc48d0#commitcomment-40301912 the author provides a fix for line 115, but now I'm hung up at line 278:
```
// Temporary FFT to pre transform the filters
      fft::fft_complex fft_engine = fft::fft_complex(d_corr_fft_size);
      memset(fft_engine.get_inbuf(), 0, sizeof(gr_complex) * d_corr_fft_size);
```
I'm not certain how to resolve this without creating the `fft::fft_complex` object

schneider42, 2021-10-01:
I'm not sure when and how this got fixes, but is certainly fixed now :)

closed 2021-10-01 by schneider42