muccc/gr-iridium #48: Weird Patterns in I/Q values

ssciancalepore, 2020-07-09:
We are using the gr-iridium module with a USRP X310 and an ANT500 antenna in an open field environment. We are able to receive and decode some messages from time to time, thus we assume that the overall configuration is fine and properly working.

We logged the I/Q patterns from the function iridium_qpsk_demod_cpp_impl.cc(...), and specifically we are logging the values of the data structure d_burst_after_pll. We log only the values that pass the check at the lines 333-336 (valid IRIDIUM packets), and in particular, we keep only the values of the packets that are recognized as IRIDIUM Ring Alert packets after the processing with the iridium-toolkit.

By plotting the aforementioned I/Q values, we obtain Fig. 1 and (considering a 3-dimensional histogram) Fig. 2.

[image]

[image]

The constellation appears very noisy and it seems (from the 3d-histogram) that two symbols are more frequent than the others.

Should we assume this as normal? Do you have any suggestion on how to reduce the noise affecting the IQ parameters?

schneider42, 2020-07-12:
The unique word at the beginning of each frame is BPSK (see `UW_DL` in `iridium.h`). It is 12 symbols long. The short ring alert frames are about ~115 symbols long (including the UW).

Short calculation:
BPSK part of IRA frame: 12 symbols
Occurrence of each symbol: 12/6 = 6

DQPSK part of IRA frame: 103 symbols
Occurrence of each symbol: 103/4 = ~25

This alone makes the two BPSK symbols 19% (6 / 31) more likely to appear.

There is another factor. Almost every IRA frame ends in "PAGE(NONE)":
`IRA: i-1461224717-t1 000023418 1626285568  95%  19.911 129 DL sat:75 beam:07 pos=(+61.09/-026.18) alt=791 RAI:48 ?00 bch:25 PAGE(tmsi:0cf09551 msc_id:03) PAGE(NONE)`

"PAGE(NONE)" corresponds to `111111111111111111111111111111111111111111` (42 bits) in the demodulated IRA frame. This corresponds to 21 DQPSK symbols, alternating between two positions. If by chance these are always the same two symbols (I haven't checked) and maybe even the same as the BPSK symbols mentioned above you might have the answer to your question. You need to have a look at this part of the IRA message.

Regarding the noise: The size of ring alert frames is not fixed. The detector tries to guess based on the signal strength but will always add some extra samples at the end. This might explain the random noise in your plot.

The fact that you see a cross  and not a clean constellation is probably rooted in the fact that the signal amplitude varies between each burst. Your cross does not look very sharp though. Either your signal amplitudes are generally low, or you have a high noise floor, or maybe this is "normal". I don't know yet.

What does gr-iridium output in terms of statistics? What does "from time to time" mean? Can you share a capture so we can have a look at the spectrum?

Sec42, 2020-07-12:
I agree with schneider, the imbalance in symbols would be because of the BPSK unique word at the beginning of the signals. This includes the additional BPSK messaging(IMS) header. Additionally not only IRA contain runs of the same symbol. Off the top of my head there is at least ITL with it's 94 consecutive 0-bits in the header.

Your constellation plot looks ok to me, accounting for the fact that (as far as I remember) there is no signal level normalisation. Different spot beams will be hitting the receiver with different signal strength depending on the satellites position.

schneider42, 2020-08-31:
@ssciancalepore any progress on this?

ssciancalepore, 2020-09-01:
> @ssciancalepore any progress on this?

Hi @schneider42 we verified that two symbols out of the four are more likely to appear, and the frequency reflects your statements. We also verified that the signal strength varies with the position of the satellite w.r.t. our location. Therefore, everything is now ok from our side, and the issue can be closed. Many thanks for your help and support. Thank you!

closed 2020-09-01 by ssciancalepore

schneider42, 2021-12-05:
This issue has been cited here: https://arxiv.org/pdf/2010.05470.pdf