muccc/gr-iridium #49: Online overflow detection

marcusetchart, 2020-08-28:
Hi!

I am using gr-iridium with hackrf (iridium-extractor -D 4 examples/hackrf.conf | grep "A:OK" > output.bits) and it works right. 

Does anyone know how I can detect overflow during the online processing? 

when I see "Detector in burst squelch at xxxx", first I have to check that it is not an interference, haven't I? (issue 26) 

Thanks a lot in advanced.
BR
Marcus

schneider42, 2020-09-02:
There are two kinds of "overflows":

 - When the burst detector sees too many bursts at once (caused by interference). This is called "burst squelch".
 - When the flow graph doesn't have enough time to handle all detected bursts anymore. This is called an "overflow".

The first one happens when there are burst detected on more than 80% the available bandwidth which I think will never be the case with Iridium. The only indication that this happened is the message you mentioned above.
The second one happens when there are more than 500 bursts pending to be processed (per `gr-iridium` internal channel). Dropped bursts are counted in the `d: ` part of the statistics.

Assuming that your issue is with the burst squelch.  I'd suggest to try (some of) the following options:

 - Inspect the signal using Inspectrum, foshpor or gqrx to see what the cause of the interference might be
 - Remove the interfering signal
 - Add a filter
 - Move the antenna
 - Shield the amplifier
 - Shield the SDR

I'm happy to look at a raw IQ capture if you can provide one.

schneider42, 2020-09-02:
I have to correct myself, there is a third kind of overflow: When there is not enough CPU/memory bandwidth available to create the detected bursts. Depending on the SDR you will see `O`s being printed (rtl-sdr, hackrf) or no indication at all (bladerf). These overflows are not tracked in the statistics and can have the following causes:
 - Computer to "slow"
 - Interference causing many wrongly detected bursts (creating additional computational load)

marcusetchart, 2020-09-10:
Thank's Sec for your soon and clearly response.

I will try to take a raw IQ capture for you.

closed 2022-02-01 by schneider42