muccc/gr-iridium #50: (Request) iridium_extractor_flowgraph with gr-soapy/sdrplay settings

alphafox02, 2020-09-12:
I’ve noticed of course that the stock flow graph is using osmosdr. What’s interesting is that I can sort of get an rsp1a to work or at least recognized with the iridium extractor, however I think it’s impossible to pass across relevant gain settings etc. 

If I open the example flow graph in the examples folder and alter the file input to that of the gr-soapy (soapy source) and change it to sdrplay in the drop down and then go and save the block each time I configure a relevant setting for sdrplay it appears to work when running the flow graph within gnuradio companion. Meaning the terminal window used to start gnuradio companion starts populating with lots of number, OK mentions, and % numbers. 

I realize this is a very specific use case, but would it be possible for you to make a branch specific to soapy source/sdrplay? I suppose the ideal solution would be I just get it to work with osmosdr but I have had way better results with gr-soapy (soapy source) for sdrplay equipment  with gnuradio 3.8.

Sec42, 2020-09-12:
Hi,

iridium-extractor can read the raw sdr samples from stdin in a variety of formats (complex, c16, u8 ...). As long as your have a commandline utility similar to rtl_sdr you should be able to use any sdr as a source.

Sec42, 2022-02-07:
Soapy source have been implemented in around the end of last year (see e7af45edb5918576499e1aba3c9fdb3b96e18c92). Please let us know if it works for you.

alphafox02, 2022-02-08:
`It` has worked in my past testing. Thank you.

closed 2022-02-08 by alphafox02