muccc/gr-iridium #54: error: no matching function for call to ‘gr::filter::kernel::fir_filter

xscode, 2021-03-01:
I've been unable to compile due to the following errors, I have all required modules installed.

```
[  4%] Building CXX object lib/CMakeFiles/gnuradio-iridium.dir/burst_downmix_impl.cc.o
/home/stuart/gr-iridium/lib/burst_downmix_impl.cc: In constructor ‘gr::iridium::burst_downmix_impl::burst_downmix_impl(int, int, size_t, const std::vector<float>&, const std::vector<float>&, bool)’:
/home/stuart/gr-iridium/lib/burst_downmix_impl.cc:144:71: error: no matching function for call to ‘gr::filter::kernel::fir_filter<std::complex<float>, std::complex<float>, float>::fir_filter(const std::vector<float>&)’
  144 |               d_cfo_est_fft(d_cfo_est_fft_size * d_fft_over_size_facor)
      |                                                                       ^
In file included from /home/stuart/gr-iridium/lib/burst_downmix_impl.h:25,
                 from /home/stuart/gr-iridium/lib/burst_downmix_impl.cc:32:
/usr/include/gnuradio/filter/fir_filter.h:36:5: note: candidate: ‘gr::filter::kernel::fir_filter<IN_T, OUT_T, TAP_T>::fir_filter(gr::filter::kernel::fir_filter<IN_T, OUT_T, TAP_T>&&) [with IN_T = std::complex<float>; OUT_T = std::complex<float>; TAP_T = float]’
   36 |     fir_filter(fir_filter&&) = default;
      |     ^~~~~~~~~~
/usr/include/gnuradio/filter/fir_filter.h:36:16: note:   no known conversion for argument 1 from ‘const std::vector<float>’ to ‘gr::filter::kernel::fir_filter<std::complex<float>, std::complex<float>, float>&&’
   36 |     fir_filter(fir_filter&&) = default;
      |                ^~~~~~~~~~~~
/usr/include/gnuradio/filter/fir_filter.h:28:5: note: candidate: ‘gr::filter::kernel::fir_filter<IN_T, OUT_T, TAP_T>::fir_filter(int, const std::vector<TAP_T>&) [with IN_T = std::complex<float>; OUT_T = std::complex<float>; TAP_T = float]’
   28 |     fir_filter(int decimation, const std::vector<TAP_T>& taps);
      |     ^~~~~~~~~~
/usr/include/gnuradio/filter/fir_filter.h:28:5: note:   candidate expects 2 arguments, 1 provided
/home/stuart/gr-iridium/lib/burst_downmix_impl.cc:144:71: error: no matching function for call to ‘gr::filter::kernel::fir_filter<float, float, float>::fir_filter(const std::vector<float>&)’
  144 |               d_cfo_est_fft(d_cfo_est_fft_size * d_fft_over_size_facor)
      |                                                                       ^
In file included from /home/stuart/gr-iridium/lib/burst_downmix_impl.h:25,
                 from /home/stuart/gr-iridium/lib/burst_downmix_impl.cc:32:
/usr/include/gnuradio/filter/fir_filter.h:36:5: note: candidate: ‘gr::filter::kernel::fir_filter<IN_T, OUT_T, TAP_T>::fir_filter(gr::filter::kernel::fir_filter<IN_T, OUT_T, TAP_T>&&) [with IN_T = float; OUT_T = float; TAP_T = float]’
   36 |     fir_filter(fir_filter&&) = default;
      |     ^~~~~~~~~~
/usr/include/gnuradio/filter/fir_filter.h:36:16: note:   no known conversion for argument 1 from ‘const std::vector<float>’ to ‘gr::filter::kernel::fir_filter<float, float, float>&&’
   36 |     fir_filter(fir_filter&&) = default;
      |                ^~~~~~~~~~~~
/usr/include/gnuradio/filter/fir_filter.h:28:5: note: candidate: ‘gr::filter::kernel::fir_filter<IN_T, OUT_T, TAP_T>::fir_filter(int, const std::vector<TAP_T>&) [with IN_T = float; OUT_T = float; TAP_T = float]’
   28 |     fir_filter(int decimation, const std::vector<TAP_T>& taps);
      |     ^~~~~~~~~~
/usr/include/gnuradio/filter/fir_filter.h:28:5: note:   candidate expects 2 arguments, 1 provided
/home/stuart/gr-iridium/lib/burst_downmix_impl.cc:144:71: error: no matching function for call to ‘gr::filter::kernel::fir_filter<std::complex<float>, std::complex<float>, float>::fir_filter(std::vector<float>)’
  144 |               d_cfo_est_fft(d_cfo_est_fft_size * d_fft_over_size_facor)
      |                                                                       ^
In file included from /home/stuart/gr-iridium/lib/burst_downmix_impl.h:25,
                 from /home/stuart/gr-iridium/lib/burst_downmix_impl.cc:32:
/usr/include/gnuradio/filter/fir_filter.h:36:5: note: candidate: ‘gr::filter::kernel::fir_filter<IN_T, OUT_T, TAP_T>::fir_filter(gr::filter::kernel::fir_filter<IN_T, OUT_T, TAP_T>&&) [with IN_T = std::complex<float>; OUT_T = std::complex<float>; TAP_T = float]’
   36 |     fir_filter(fir_filter&&) = default;
      |     ^~~~~~~~~~
/usr/include/gnuradio/filter/fir_filter.h:36:16: note:   no known conversion for argument 1 from ‘std::vector<float>’ to ‘gr::filter::kernel::fir_filter<std::complex<float>, std::complex<float>, float>&&’
   36 |     fir_filter(fir_filter&&) = default;
      |                ^~~~~~~~~~~~
/usr/include/gnuradio/filter/fir_filter.h:28:5: note: candidate: ‘gr::filter::kernel::fir_filter<IN_T, OUT_T, TAP_T>::fir_filter(int, const std::vector<TAP_T>&) [with IN_T = std::complex<float>; OUT_T = std::complex<float>; TAP_T = float]’
   28 |     fir_filter(int decimation, const std::vector<TAP_T>& taps);
      |     ^~~~~~~~~~
/usr/include/gnuradio/filter/fir_filter.h:28:5: note:   candidate expects 2 arguments, 1 provided
/home/stuart/gr-iridium/lib/burst_downmix_impl.cc:144:71: error: no matching function for call to ‘gr::filter::kernel::fir_filter<std::complex<float>, std::complex<float>, float>::fir_filter(std::vector<float>)’
  144 |               d_cfo_est_fft(d_cfo_est_fft_size * d_fft_over_size_facor)
      |                                                                       ^
In file included from /home/stuart/gr-iridium/lib/burst_downmix_impl.h:25,
                 from /home/stuart/gr-iridium/lib/burst_downmix_impl.cc:32:
/usr/include/gnuradio/filter/fir_filter.h:36:5: note: candidate: ‘gr::filter::kernel::fir_filter<IN_T, OUT_T, TAP_T>::fir_filter(gr::filter::kernel::fir_filter<IN_T, OUT_T, TAP_T>&&) [with IN_T = std::complex<float>; OUT_T = std::complex<float>; TAP_T = float]’
   36 |     fir_filter(fir_filter&&) = default;
      |     ^~~~~~~~~~
/usr/include/gnuradio/filter/fir_filter.h:36:16: note:   no known conversion for argument 1 from ‘std::vector<float>’ to ‘gr::filter::kernel::fir_filter<std::complex<float>, std::complex<float>, float>&&’
   36 |     fir_filter(fir_filter&&) = default;
      |                ^~~~~~~~~~~~
/usr/include/gnuradio/filter/fir_filter.h:28:5: note: candidate: ‘gr::filter::kernel::fir_filter<IN_T, OUT_T, TAP_T>::fir_filter(int, const std::vector<TAP_T>&) [with IN_T = std::complex<float>; OUT_T = std::complex<float>; TAP_T = float]’
   28 |     fir_filter(int decimation, const std::vector<TAP_T>& taps);
      |     ^~~~~~~~~~
/usr/include/gnuradio/filter/fir_filter.h:28:5: note:   candidate expects 2 arguments, 1 provided
/home/stuart/gr-iridium/lib/burst_downmix_impl.cc: In member function ‘virtual int gr::iridium::burst_downmix_impl::work(int, gr_vector_const_void_star&, gr_vector_void_star&)’:
/home/stuart/gr-iridium/lib/burst_downmix_impl.cc:833:5: warning: no return statement in function returning non-void [-Wreturn-type]
  833 |     }
      |     ^
make[2]: *** [lib/CMakeFiles/gnuradio-iridium.dir/build.make:102: lib/CMakeFiles/gnuradio-iridium.dir/burst_downmix_impl.cc.o] Error 1
make[2]: Leaving directory '/home/stuart/gr-iridium/build'
make[1]: *** [CMakeFiles/Makefile2:249: lib/CMakeFiles/gnuradio-iridium.dir/all] Error 2
make[1]: Leaving directory '/home/stuart/gr-iridium/build'
make: *** [Makefile:141: all] Error 2
```

schneider42, 2021-03-01:
Which version of GNU Radio is installed on your system?

schneider42, 2021-03-03:
Strange. The 3.9.0.0 tag should have the updated constructor available: https://github.com/gnuradio/gnuradio/commit/0c62af72fe30c0a9009fe7cbb6de2efc853b7d37

I'll have to dig deeper

schneider42, 2021-03-03:
I can reproduce this issue and raised a related question on the GNURadio mailing list.

schneider42, 2021-03-15:
Which PPA did you configure?

I realized that `ppa:gnuradio/gnuradio-releases` works but `ppa:gnuradio/gnuradio-master` is in an undefined state but still identifies as 3.9.0.0.

schneider42, 2021-10-01:
Solution is simply: Do not use `ppa:gnuradio/gnuradio-master`

closed 2021-10-01 by schneider42