muccc/gr-iridium #56: Getting ERROR ON RUN TIME

justin983, 2021-03-25:
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.1.0
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
[INFO] [UHD] linux; GNU C++ version 9.2.1 20200304; Boost_107100; UHD_3.15.0.0-2build5
Detached kernel driver
Found Rafael Micro R820T tuner
Reattached kernel driver
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Detached kernel driver
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 2000000.052982 Hz
[R82XX] PLL not locked!
(RF) Gain: 19.7 (Requested 19)
WARNING: Gain if not supported by source!
Supported gains: ('LNA',)
Bandwidth: 0.0 (Requested 0)
Warning: Setting antenna to RX
1616693210 | i:   0/s | i_avg:   0/s | q:    0 | q_max:    0 | o:    0/s | ok:   0% | ok:   0/s | ok_avg:   0% | ok:          0 | ok_avg:   0/s | d: 0
Traceback (most recent call last):
  File "/usr/local/bin/iridium-extractor", line 247, in <module>
    tb.run()
  File "/usr/local/lib/python3/dist-packages/gnuradio/gr/top_block.py", line 123, in run
    self.start(max_noutput_items)
  File "/usr/local/lib/python3/dist-packages/gnuradio/gr/top_block.py", line 111, in start
    top_block_start_unlocked(self._impl, max_noutput_items)
  File "/usr/local/lib/python3/dist-packages/gnuradio/gr/runtime_swig.py", line 4832, in top_block_start_unlocked
    return _runtime_swig.top_block_start_unlocked(r, max_noutput_items)
RuntimeError: list contains invalid format!

**Getting Above error on both RTL-SDR AND HACKRF**
**gnuradio 3.8.1.0 is already build and installed** 
**please help me out**

schneider42, 2021-03-26:
Can you detail how you installed GNU Radio?

schneider42, 2021-03-26:
Also: Which branch of gr-iridium are you using? Did you try the `maint-3.8` branch?

justin983, 2021-03-26:
@schneider42 
yes i maint-3.8 branch

justin983, 2021-03-26:
@schneider42 

sudo apt install git cmake g++ libboost-all-dev libgmp-dev swig python3-numpy python3-mako python3-sphinx python3-lxml doxygen libfftw3-dev libsdl1.2-dev libgsl-dev libqwt-qt5-dev libqt5opengl5-dev python3-pyqt5 liblog4cpp5-dev libzmq3-dev python3-yaml python3-click python3-click-plugins python3-zmq python3-scipy python3-pip python3-gi-cairo
pip3 install git+https://github.com/pyqtgraph/pyqtgraph@develop
pip3 install numpy scipy
echo 'export PYTHONPATH=/usr/local/lib/python3/dist-packages:usr/local/lib/python2.7/site-packages:$PYTHONPATH' >> ~/.bashrc
echo 'export LD_LIBRARY_PATH=/user/local/lib:$LD_LIBRARY_PATH' >> ~/.bashrc
echo 'export PYTHONPATH=/usr/local/lib/python3/dist-packages:usr/local/lib/python2.7/site-packages:$PYTHONPATH' >> ~/.profile
echo 'export LD_LIBRARY_PATH=/user/local/lib:$LD_LIBRARY_PATH' >> ~/.profile
cd ~/
git clone --recursive https://github.com/gnuradio/gnuradio
cd gnuradio
git checkout maint-3.8
mkdir build
cd build
git pull --recurse-submodules=on
git submodule update --init
cmake -DENABLE_GR_UHD=OFF ..
make -j $(nproc --all)
sudo make install
sudo ldconfig

justin983, 2021-03-26:
@schneider42 
i just did a NEW UBUNTU-64 BIT INSTALL 
and build gnuradio 3.10.* which is successful 

BUT WHILE RUNNING gr-iridium 
getting this error 

  File "/usr/local/bin/iridium-extractor", line 232, in <module>
    tb = iridium.iridium_extractor_flowgraph.FlowGraph(center_frequency=center, sample_rate=sample_rate, decimation=decimation, 
  File "/usr/local/lib/python3/dist-packages/iridium/iridium_extractor_flowgraph.py", line 244, in __init__
    tb.connect(source, self._fft_burst_tagger)
  File "/usr/local/lib/python3/dist-packages/gnuradio/gr/hier_block2.py", line 37, in wrapped
    func(self, src, src_port, dst, dst_port)
  File "/usr/local/lib/python3/dist-packages/gnuradio/gr/hier_block2.py", line 100, in connect
    self.primitive_connect(*args)
TypeError: primitive_connect(): incompatible function arguments. The following argument types are supported:
    1. (self: gnuradio.gr.gr_python.hier_block2_pb, block: gnuradio.gr.gr_python.basic_block) -> None
    2. (self: gnuradio.gr.gr_python.hier_block2_pb, src: gnuradio.gr.gr_python.basic_block, src_port: int, dst: gnuradio.gr.gr_python.basic_block, dst_port: int) -> None

Invoked with: <gnuradio.gr.gr_python.top_block_pb object at 0x7fd45a65f170>, <Swig Object of type 'gr::basic_block_sptr *' at 0x7fd455b37180>, 0, <gnuradio.gr.gr_python.basic_block object at 0x7fd45a08b070>, 0
Tagged 0 bursts

Sec42, 2021-06-26:
If you're still stuck on this: can you try gnuradio-3.9 with the maint-3.9 branch?  I just tested that combination and it worked fine for me.

schneider42, 2021-10-01:
I consider gnuradio 3.10 not supported by this OOT. Please install gnuradio 3.9 as described here: https://github.com/muccc/gr-iridium/blob/master/README.md#gnuradio-39-on-ubuntu-or-debian