muccc/gr-iridium #57: ERROR on gr-iridium with gnuradio 3.10.* on run time

justin983, 2021-03-26:
i just did a NEW UBUNTU-64 BIT INSTALL 
and build gnuradio 3.10.* which is successful 

BUT WHILE RUNNING gr-iridium 
getting this error 

  File "/usr/local/bin/iridium-extractor", line 232, in <module>
    tb = iridium.iridium_extractor_flowgraph.FlowGraph(center_frequency=center, sample_rate=sample_rate, decimation=decimation, 
  File "/usr/local/lib/python3/dist-packages/iridium/iridium_extractor_flowgraph.py", line 244, in __init__
    tb.connect(source, self._fft_burst_tagger)
  File "/usr/local/lib/python3/dist-packages/gnuradio/gr/hier_block2.py", line 37, in wrapped
    func(self, src, src_port, dst, dst_port)
  File "/usr/local/lib/python3/dist-packages/gnuradio/gr/hier_block2.py", line 100, in connect
    self.primitive_connect(*args)
TypeError: primitive_connect(): incompatible function arguments. The following argument types are supported:
    1. (self: gnuradio.gr.gr_python.hier_block2_pb, block: gnuradio.gr.gr_python.basic_block) -> None
    2. (self: gnuradio.gr.gr_python.hier_block2_pb, src: gnuradio.gr.gr_python.basic_block, src_port: int, dst: gnuradio.gr.gr_python.basic_block, dst_port: int) -> None

Invoked with: <gnuradio.gr.gr_python.top_block_pb object at 0x7fd45a65f170>, <Swig Object of type 'gr::basic_block_sptr *' at 0x7fd455b37180>, 0, <gnuradio.gr.gr_python.basic_block object at 0x7fd45a08b070>, 0
Tagged 0 bursts

schneider42, 2021-03-26:
The last line of your output has the term "Swig" inside. SWIG was dropped from GR in version 3.9. I'm wondering why it still mentions this here. You are using the master branch of gr-iridium now? This is all very strange...

justin983, 2021-03-27:
@schneider42 
yes i used master branch of gr-iridium (https://github.com/muccc/gr-iridium.git) this one.

Should I try with GnuRadio 3.9 ?

justin983, 2021-03-27:
@schneider42 
I also tried with gr-iridium 3.9 maint but getting same error!

robinjanssens, 2021-04-20:
I replicated this issue on a fresh Ubuntu 20.04.2 install.
`apt list | grep gnuradio

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

gnuradio-dev/focal,now 3.10.0.git~master~14409~focal-1 amd64 [installed,automatic]
gnuradio-doc/focal,focal 3.10.0.git~master~14409~focal-1 all
gnuradio/focal,now 3.10.0.git~master~14409~focal-1 amd64 [installed]
libgnuradio-analog3.8.1/focal,now 3.8.1.0~rc1-2build2 amd64 [installed,automatic]
libgnuradio-analog/focal 3.9.0.0git~master~14059~focal-4 amd64
libgnuradio-audio3.8.1/focal,now 3.8.1.0~rc1-2build2 amd64 [installed,automatic]
libgnuradio-audio/focal 3.9.0.0git~master~14059~focal-4 amd64
libgnuradio-blocks3.8.1/focal,now 3.8.1.0~rc1-2build2 amd64 [installed,automatic]
libgnuradio-blocks/focal 3.9.0.0git~master~14059~focal-4 amd64
libgnuradio-channels3.8.1/focal 3.8.1.0~rc1-2build2 amd64
libgnuradio-channels/focal 3.9.0.0git~master~14059~focal-4 amd64
libgnuradio-dab3.8.0/focal 0.4-2build3 amd64
libgnuradio-digital3.8.1/focal,now 3.8.1.0~rc1-2build2 amd64 [installed,automatic]
libgnuradio-digital/focal 3.9.0.0git~master~14059~focal-4 amd64
libgnuradio-dtv3.8.1/focal 3.8.1.0~rc1-2build2 amd64
libgnuradio-dtv/focal 3.9.0.0git~master~14059~focal-4 amd64
libgnuradio-fcdproplus3.8.0/focal,now 3.8~20190817-3build3 amd64 [installed,automatic]
libgnuradio-fec3.8.1/focal 3.8.1.0~rc1-2build2 amd64
libgnuradio-fec/focal 3.9.0.0git~master~14059~focal-4 amd64
libgnuradio-fft3.8.1/focal,now 3.8.1.0~rc1-2build2 amd64 [installed,automatic]
libgnuradio-fft/focal 3.9.0.0git~master~14059~focal-4 amd64
libgnuradio-filter3.8.1/focal,now 3.8.1.0~rc1-2build2 amd64 [installed,automatic]
libgnuradio-filter/focal 3.9.0.0git~master~14059~focal-4 amd64
libgnuradio-fosphor3.8.0/focal,now 3.8~2.2d4fe78-1build3 amd64 [installed,automatic]
libgnuradio-hpsdr1.2.1/focal 1.2.1-1 amd64
libgnuradio-iio1/focal 0.3-7build3 amd64
libgnuradio-iqbalance3.8.0/focal,now 0.38-4build3 amd64 [installed,automatic]
libgnuradio-limesdr3.0.1/focal 3.0.1-2build1 amd64
libgnuradio-network/focal 3.9.0.0git~master~14059~focal-4 amd64
libgnuradio-osmosdr0.2.0/focal,now 0.2.0-2 amd64 [installed,automatic]
libgnuradio-pmt3.8.1/focal,now 3.8.1.0~rc1-2build2 amd64 [installed,automatic]
libgnuradio-pmt/focal 3.9.0.0git~master~14059~focal-4 amd64
libgnuradio-qtgui3.8.1/focal 3.8.1.0~rc1-2build2 amd64
libgnuradio-qtgui/focal 3.9.0.0git~master~14059~focal-4 amd64
libgnuradio-radar1.0.0/focal 0.0.0.20180308-4build3 amd64
libgnuradio-rds1/focal 3.8.0.0.f1c584a-3build3 amd64
libgnuradio-runtime3.8.1/focal,now 3.8.1.0~rc1-2build2 amd64 [installed,automatic]
libgnuradio-runtime/focal 3.9.0.0git~master~14059~focal-4 amd64
libgnuradio-trellis3.8.1/focal 3.8.1.0~rc1-2build2 amd64
libgnuradio-trellis/focal 3.9.0.0git~master~14059~focal-4 amd64
libgnuradio-uhd3.8.1/focal,now 3.8.1.0~rc1-2build2 amd64 [installed,automatic]
libgnuradio-uhd/focal 3.9.0.0git~master~14059~focal-4 amd64
libgnuradio-video-sdl3.8.1/focal 3.8.1.0~rc1-2build2 amd64
libgnuradio-video-sdl/focal 3.9.0.0git~master~14059~focal-4 amd64
libgnuradio-vocoder3.8.1/focal 3.8.1.0~rc1-2build2 amd64
libgnuradio-vocoder/focal 3.9.0.0git~master~14059~focal-4 amd64
libgnuradio-wavelet3.8.1/focal 3.8.1.0~rc1-2build2 amd64
libgnuradio-wavelet/focal 3.9.0.0git~master~14059~focal-4 amd64
libgnuradio-zeromq3.8.1/focal 3.8.1.0~rc1-2build2 amd64
libgnuradio-zeromq/focal 3.9.0.0git~master~14059~focal-4 amd64`

Although GNURadio 3.10 is installed it seem to run using GNURadio 3.8

`iridium-extractor -D 4 examples/rtl-sdr.conf  > ../output.bits
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.1.0
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
[INFO] [UHD] linux; GNU C++ version 9.2.1 20200304; Boost_107100; UHD_3.15.0.0-2build5
Found Rafael Micro R820T tuner
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 2000000.052982 Hz
[R82XX] PLL not locked!
(RF) Gain: 19.7 (Requested 19)
WARNING: Gain if not supported by source!
Supported gains: ('LNA',)
Bandwidth: 0.0 (Requested 0)
Warning: Setting antenna to RX
Traceback (most recent call last):
  File "/usr/local/bin/iridium-extractor", line 244, in <module>
    tb = iridium.iridium_extractor_flowgraph.FlowGraph(center_frequency=center, sample_rate=sample_rate, decimation=decimation, 
  File "/usr/local/lib/python3/dist-packages/iridium/iridium_extractor_flowgraph.py", line 248, in __init__
    tb.connect(source, self._fft_burst_tagger)
  File "/usr/lib/python3/dist-packages/gnuradio/gr/hier_block2.py", line 37, in wrapped
    func(self, src, src_port, dst, dst_port)
  File "/usr/lib/python3/dist-packages/gnuradio/gr/hier_block2.py", line 100, in connect
    self.primitive_connect(*args)
TypeError: primitive_connect(): incompatible function arguments. The following argument types are supported:
    1. (self: gnuradio.gr.gr_python.hier_block2_pb, block: gnuradio.gr.gr_python.basic_block) -> None
    2. (self: gnuradio.gr.gr_python.hier_block2_pb, src: gnuradio.gr.gr_python.basic_block, src_port: int, dst: gnuradio.gr.gr_python.basic_block, dst_port: int) -> None

Invoked with: <gnuradio.gr.gr_python.top_block_pb object at 0x7fbb14c8bcf0>, <Swig Object of type 'gr::basic_block_sptr *' at 0x7fbb10018150>, 0, <gnuradio.gr.gr_python.basic_block object at 0x7fbb14575f30>, 0
Tagged 0 bursts
`
Is it possible that this is caused by gr-osmosdr?

Sec42, 2021-04-20:
Where did you get this "3.10" package from?

As far as I can tell, ubuntu 20.4 only ships with gnuradio 3.8
at least according to https://packages.ubuntu.com/focal/gnuradio

Also all your other gnuradio packages are installed in both 3.8 and 3.9 version.

To me it does not look like a clean/fresh install - did you add some external ppa and/or install a mix of packages from different sources?

robinjanssens, 2021-04-20:
Yes, I got them by adding the latest PPA as described in the docs
`sudo add-apt-repository ppa:gnuradio/gnuradio-master`
[https://wiki.gnuradio.org/index.php/InstallingGR#Ubuntu_PPA_Installation](https://wiki.gnuradio.org/index.php/InstallingGR#Ubuntu_PPA_Installation)

schneider42, 2021-04-21:
Please do not use the `gnuradio-master` PPA. In my experience it's contents are undefined. Please try the `gnuradio-releases` PPA.

schneider42, 2021-10-01:
I consider GNURadio 3.10 unsupported by this OOT for now. Please do not use `ppa:gnuradio/gnuradio-master` and wait for GNURadio 3.10 to be officially released via `ppa:gnuradio/gnuradio-releases`.

Please also note that you **MUST NOT** install any other OOT modules like `gr-osmosdr` via your package manager. They need to be compiled from source after installing that PPA. In fact the only way to install GNURadio 3.9 (on Ubuntu) at the moment that I can recommend is described here: https://github.com/muccc/gr-iridium/issues/62#issuecomment-891291015

Everything else seems to be in some way broken or not documented.

schneider42, 2021-10-01:
I documented this here: https://github.com/muccc/gr-iridium/blob/master/README.md#gnuradio-39-on-ubuntu-or-debian

closed 2021-10-01 by schneider42