muccc/gr-iridium #59: GR-iridium install issue

BrunoKlu, 2021-06-01:
Hello guys,

I'm having this error with the cmake command 
`nobsklu@nobsklu-HP-Elite-x2-1012-G1:~/Desktop/gr-iridium2/build$ cmake ..
-- Build type not specified: defaulting to release.
-- Checking for module 'mpir >= 3.0'
--   No package 'mpir' found
-- Could NOT find MPIR (missing: MPIRXX_LIBRARY MPIR_INCLUDE_DIR) 
-- User set python executable /usr/bin/python3
CMake Error at CMakeLists.txt:128 (find_package):
  By not providing "Findpybind11.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "pybind11",
  but CMake did not find one.

  Could not find a package configuration file provided by "pybind11" with any
  of the following names:

    pybind11Config.cmake
    pybind11-config.cmake

  Add the installation prefix of "pybind11" to CMAKE_PREFIX_PATH or set
  "pybind11_DIR" to a directory containing one of the above files.  If
  "pybind11" provides a separate development package or SDK, be sure it has
  been installed.

-- Configuring incomplete, errors occurred!
See also "/home/nobsklu/Desktop/gr-iridium2/build/CMakeFiles/CMakeOutput.log".
See also "/home/nobsklu/Desktop/gr-iridium2/build/CMakeFiles/CMakeError.log".`

Could Someone help me with this please ?

Many thanks

BrunoKlu, 2021-06-01:
[CMakeError.log](https://github.com/muccc/gr-iridium/files/6578317/CMakeError.log)
[CMakeOutput.log](https://github.com/muccc/gr-iridium/files/6578318/CMakeOutput.log)

Sec42, 2021-07-26:
Not sure if that's still relevant: you need to use the correct branch depending on your installed gnuradio version. (maint-3.8 or maint-3.9).

For gnuradio-3.9 you also need to install pybind11 - i.e. for ubuntu "sudo apt install pybind11-dev"

schneider42, 2022-02-01:
@BrunoKlu are you still experiencing this issue?

closed 2022-03-25 by schneider42