muccc/gr-iridium #60: can not run iridium extractor

Mahmoudshakra, 2021-06-16:
Hi all , 
 please help me as i can not run iridium extractor ? i am wondering why ? 

1623837871 | i:   0/s | i_avg:   0/s | q:    0 | q_max:    0 | o:    0/s | ok:   0% | ok:   0/s | ok_avg:   0% | ok:          0 | ok_avg:   0/s | d: 0
Traceback (most recent call last):
  File "/usr/local/bin/iridium-extractor", line 247, in <module>
    tb.run()
  File "/usr/lib/python3/dist-packages/gnuradio/gr/top_block.py", line 123, in run
    self.start(max_noutput_items)
  File "/usr/lib/python3/dist-packages/gnuradio/gr/top_block.py", line 111, in start
    top_block_start_unlocked(self._impl, max_noutput_items)
  File "/usr/lib/python3/dist-packages/gnuradio/gr/runtime_swig.py", line 4832, in top_block_start_unlocked
    return _runtime_swig.top_block_start_unlocked(r, max_noutput_items)
RuntimeError: list contains invalid format!

Sec42, 2021-06-16:
Hard to tell as you gave no further information

- which OS are you running
- which iridium-toolkit version did you check out?
- which python version are you using
- which gnuradio version is installed and how did you install it

schneider42, 2022-02-01:
@Mahmoudshakra are you still interested in this? Otherwise I'm going to close the issue.

closed 2022-03-25 by schneider42