muccc/gr-iridium #61: Build error with gcc-11

cyberpunkrocker-zero, 2021-06-18:
I'm trying to build gr-iridium on Arch Linux (gcc-11.1.0, gnuradio-3.8.2.0), but I'm getting an error which, I suppose, is caused by the stricter requirements of gcc-11. Quite a lot of warnings & errors from doxygen, too. Also, you have not specified the python version gr-iridium needs... I assume python-3.x is OK?

```
Branch 'maint-3.8' set up to track remote branch 'maint-3.8' from 'origin'.
Switched to a new branch 'maint-3.8'
-- The CXX compiler identification is GNU 11.1.0
-- The C compiler identification is GNU 11.1.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Build type not specified: defaulting to release.
-- Found LOG4CPP: /usr/lib/liblog4cpp.so
-- Found PkgConfig: /usr/bin/pkg-config (found version "1.7.3") 
-- Checking for module 'gmp'
--   Found gmp, version 6.2.1
-- Found GMP: /usr/lib/libgmpxx.so  
-- Checking for module 'mpir >= 3.0'
--   Package 'mpir', required by 'virtual:world', not found
-- Could NOT find MPIR (missing: MPIRXX_LIBRARY MPIR_LIBRARY MPIR_INCLUDE_DIR) 
-- Found MPLIB: /usr/lib/libgmpxx.so  
-- Found Boost: /usr/lib64/cmake/Boost-1.76.0/BoostConfig.cmake (found suitable version "1.76.0", minimum required is "1.76.0") found components: date_time program_options filesystem system regex thread unit_test_framework 
-- Found Volk: Volk::volk  
-- User set python executable /usr/bin/python3
-- Found PythonInterp: /usr/bin/python3 (found version "3.9.5") 
-- Found PythonLibs: /usr/lib/libpython3.9.so (found suitable exact version "3.9.5") 
-- Checking for module 'fftw3f >= 3.0'
--   Found fftw3f , version 3.3.9
-- Found FFTW3f: /lib/libfftw3f.so  
-- Found Git: /usr/bin/git  
-- Extracting version information from git describe...
-- Found Doxygen: /usr/bin/doxygen (found version "1.9.1") found components: doxygen dot 
-- Using install prefix: /usr
-- Building for version: 48cdeafd / 1.0.0git
-- No C++ unit tests... skipping
-- 
-- Checking for module SWIG
-- Found SWIG version 4.0.2.
-- Found SWIG: /usr/bin/swig  
-- Found PythonLibs: /usr/lib/libpython3.9.so (found version "3.9.5") 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/madmax/aur/gr-iridium-git/src/gr-iridium/build
[  4%] Building CXX object swig/CMakeFiles/_iridium_swig_doc_tag.dir/_iridium_swig_doc_tag.cpp.o
[  9%] Shebangin iridium-extractor
[ 14%] Generating documentation with doxygen
[ 19%] Building CXX object lib/CMakeFiles/gnuradio-iridium.dir/fft_burst_tagger_impl.cc.o
[ 33%] Building CXX object lib/CMakeFiles/gnuradio-iridium.dir/burst_downmix_impl.cc.o
[ 33%] Generating __init__.pyc, iridium_extractor_flowgraph.pyc, iridium_qpsk_demod.pyc, gr_iridium.pyc, gr_iridium_demod.pyc, iridium_frame_printer.pyc, frame_sorter.pyc
[ 33%] Building CXX object lib/CMakeFiles/gnuradio-iridium.dir/iuchar_to_complex_impl.cc.o
[ 42%] Building CXX object lib/CMakeFiles/gnuradio-iridium.dir/pdu_round_robin_impl.cc.o
[ 42%] Building CXX object lib/CMakeFiles/gnuradio-iridium.dir/tagged_burst_to_pdu_impl.cc.o
[ 47%] Generating __init__.pyo, iridium_extractor_flowgraph.pyo, iridium_qpsk_demod.pyo, gr_iridium.pyo, gr_iridium_demod.pyo, iridium_frame_printer.pyo, frame_sorter.pyo
[ 52%] Building CXX object lib/CMakeFiles/gnuradio-iridium.dir/iridium_qpsk_demod_cpp_impl.cc.o
[ 57%] Building CXX object lib/CMakeFiles/gnuradio-iridium.dir/pdu_null_sink_impl.cc.o
warning: Tag 'TCL_SUBST' at line 207 of file '/home/madmax/aur/gr-iridium-git/src/gr-iridium/build/docs/doxygen/Doxyfile' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Invalid value for 'FILTER_PATTERNS' tag at line 793, file /home/madmax/aur/gr-iridium-git/src/gr-iridium/build/docs/doxygen/Doxyfile: Values in list '*.py="/home/madmax/aur/gr-iridium-git/src/gr-iridium"/doc/doxygen/other/doxypy.py' are not properly space or comma separated
warning: Tag 'COLS_IN_ALPHA_INDEX' at line 886 of file '/home/madmax/aur/gr-iridium-git/src/gr-iridium/build/docs/doxygen/Doxyfile' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Tag 'PERL_PATH' at line 1686 of file '/home/madmax/aur/gr-iridium-git/src/gr-iridium/build/docs/doxygen/Doxyfile' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Tag 'MSCGEN_PATH' at line 1707 of file '/home/madmax/aur/gr-iridium-git/src/gr-iridium/build/docs/doxygen/Doxyfile' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
[... 395 lines cut ...]
In file included from /usr/include/boost/mpl/aux_/include_preprocessed.hpp:37,
                 from /usr/include/boost/mpl/placeholders.hpp:43,
                 from /usr/include/boost/iterator/iterator_categories.hpp:16,
                 from /usr/include/boost/iterator/iterator_adaptor.hpp:14,
                 from /usr/include/boost/iterator/reverse_iterator.hpp:10,
                 from /usr/include/boost/range/reverse_iterator.hpp:21,
                 from /usr/include/boost/range/rend.hpp:19,
                 from /usr/include/boost/foreach.hpp:82,
                 from /usr/include/gnuradio/basic_block.h:33,
                 from /usr/include/gnuradio/block.h:27,
                 from /usr/include/gnuradio/sync_block.h:27,
                 from /home/madmax/aur/gr-iridium-git/src/gr-iridium/lib/../include/iridium/tagged_burst_to_pdu.h:25,
                 from /home/madmax/aur/gr-iridium-git/src/gr-iridium/lib/tagged_burst_to_pdu_impl.h:24,
                 from /home/madmax/aur/gr-iridium-git/src/gr-iridium/lib/tagged_burst_to_pdu_impl.cc:26:
/usr/include/boost/mpl/aux_/preprocessed/gcc/placeholders.hpp:29:16: note:   'mpl_::_1'
   29 | typedef arg<1> _1;
      |                ^~
make[2]: *** [lib/CMakeFiles/gnuradio-iridium.dir/build.make:160: lib/CMakeFiles/gnuradio-iridium.dir/pdu_round_robin_impl.cc.o] Error 1
make[2]: *** [lib/CMakeFiles/gnuradio-iridium.dir/build.make:146: lib/CMakeFiles/gnuradio-iridium.dir/iridium_qpsk_demod_cpp_impl.cc.o] Error 1
make[2]: *** [lib/CMakeFiles/gnuradio-iridium.dir/build.make:118: lib/CMakeFiles/gnuradio-iridium.dir/burst_downmix_impl.cc.o] Error 1
make[2]: *** [lib/CMakeFiles/gnuradio-iridium.dir/build.make:104: lib/CMakeFiles/gnuradio-iridium.dir/tagged_burst_to_pdu_impl.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:254: lib/CMakeFiles/gnuradio-iridium.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
    Aborting...
```

Sec42, 2021-06-26:
Python 3.x should be fine.

Can you try with gnuradio 3.9 and the maint-3.9 branch? To me it looks more like issues outside of our code/control.

cyberpunkrocker-zero, 2021-07-25:
Sorry for the delay... Now that Arch Linux officially upgraded gnuradio to 3.9.2.0, I tried compiling it again, this time getting this error: 
```
CMake Error at swig/CMakeLists.txt:37 (include):
  include could not find requested file:

    GrSwig

CMake Error at swig/CMakeLists.txt:51 (GR_SWIG_MAKE):
  Unknown CMake command "GR_SWIG_MAKE".
```
And I'm getting the same erron on almost every gnuradio plugin I'm trying to (re)compile. It seems that gnuradio-3.9 dropped the use of swig in favor of pybind11, so very few older plugins work with 3.9 now. I think I have to go back to 3.8 for the time being to get a working system...

Sec42, 2021-07-26:
Yes, gnuradio modules have to be updated to work with 3.9 -- the gr-iridium has different branches for this exact reason. For use with gnuradio 3.9 please use the "maint-3.9" branch.

schneider42, 2021-09-29:
This should be solved now with 1329a6e7a9e993dc63c8a573c65b1472a382e486. Solution is to use C++ lambdas as described here: https://wiki.gnuradio.org/index.php/GNU_Radio_3.9_OOT_Module_Porting_Guide#C.2B.2B_Modernization

Also fixed this for the `maint-3.8` branch

closed 2021-09-29 by schneider42