muccc/gr-iridium #62: Error on iridium-extractor running fresh ubuntu installation

65745256426, 2021-06-30:
user@user-VirtualBox:~/Documents/gr-iridium$ iridium-extractor -D 4 examples/rtl-sdr.conf | grep "A:OK" > output.bits
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.1.0
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Detached kernel driver
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 2000000.052982 Hz
[R82XX] PLL not locked!
(RF) Gain: 19.7 (Requested 19)
WARNING: Gain if not supported by source!
Supported gains: ('LNA',)
Bandwidth: 0.0 (Requested 0)
Warning: Setting antenna to RX
Traceback (most recent call last):
  File "/usr/local/bin/iridium-extractor", line 244, in <module>
    tb = iridium.iridium_extractor_flowgraph.FlowGraph(center_frequency=center, sample_rate=sample_rate, decimation=decimation, 
  File "/usr/local/lib/python3/dist-packages/iridium/iridium_extractor_flowgraph.py", line 248, in __init__
    tb.connect(source, self._fft_burst_tagger)
  File "/usr/lib/python3/dist-packages/gnuradio/gr/hier_block2.py", line 37, in wrapped
    func(self, src, src_port, dst, dst_port)
  File "/usr/lib/python3/dist-packages/gnuradio/gr/hier_block2.py", line 100, in connect
    self.primitive_connect(*args)
TypeError: primitive_connect(): incompatible function arguments. The following argument types are supported:
    1. (self: gnuradio.gr.gr_python.hier_block2_pb, block: gnuradio.gr.gr_python.basic_block) -> None
    2. (self: gnuradio.gr.gr_python.hier_block2_pb, src: gnuradio.gr.gr_python.basic_block, src_port: int, dst: gnuradio.gr.gr_python.basic_block, dst_port: int) -> None
 
Invoked with: <gnuradio.gr.gr_python.top_block_pb object at 0x7f478a079ef0>, <Swig Object of type 'gr::basic_block_sptr *' at 0x7f4789900210>, 0, <gnuradio.gr.gr_python.basic_block object at 0x7f47899c4270>, 0
Tagged 0 bursts

I have gnuradio installed version 3.9.2.0
[image]

(Note I also had to install osmosdr manually (apt-get install gr-osmosdr) and I also encountered the issue where I enter the python path command as described in issue 5.  (Module iridium not found).

 This is a fresh ubuntu installation and I installed gnuradio from the gnuradio-releases repository as instructed in other issues)

65745256426, 2021-06-30:
Ubuntu version was 20.04.2.0

schneider42, 2021-06-30:
You mention that you have Gnuradio "3.9.2.0" installed. The output of the script mentions version "3.8.1.0" though. Are you sure there are no Gnuradio 3.8 files installed on that computer?

65745256426, 2021-06-30:
Well I am not entirely sure
 sudo add-apt-repository ppa:gnuradio/gnuradio-releases
I installed GNURadio from the releases repo using that command and then apt-get update as doing the default command of apt-get install gnuradio would install the older 3.8 version

The command gnuradio-config-info --version outputs 3.9.2.0

I then received another error saying that the library "iridium" was not found. So I proceeded to type this command
export PYTHONPATH=/usr/local/lib/python3/dist-packages
Which solved the issue but then brought about another one
 ImportError: No module named osmosdr

So I went ahead and typed 
sudo apt-get install gr-osmosdr

But now I'm receiving that other error that I've put as a screenshot. 

I'm sorry if I've made a beginners mistake here and I'm wasting everyone's time,  I found this project from a Youtube tutorial and did not encounter any of these sorts of errors that they encountered so perhaps I've just done something fundamentally wrong ?

schneider42, 2021-06-30:
The main issue here is the version hell that gnuradio is putting us through :/ This is not at all a beginners mistake.

We will have to look into what Ubuntu really does and try to reproduce your results. Might take some time.... I'll have to ask for a bit of patience until this can be figured out.

schneider42, 2021-07-31:
The problem is that `gr-osmosdr` is installed from the Ubuntu 20.04 repositories. It depends on gnuradio-3.8 and when running `apt install gr-osmosdr` a gnuradio 3.8 installation is pulled in together with it. This causes the master branch of `gr-iridium` to not work anymore as the system now has a mix of gnuradio 3.8 and 3.9 installed.

Currently there are two options for you:
 - Do not install gnuradio 3.9 from the gnuradio ppa. Install gnuradio 3.8 from Ubuntu repositories and then use the `maint-3.8` branch of `gr-iridium`
 - Install gnuradio 3.9 from the gnuradio ppa. Do not install `gr-osmosdr` from the Ubuntu repositories. Build and install `gr-osmosdr` as described here: https://osmocom.org/projects/gr-osmosdr/wiki

Sec42, 2021-08-02:
While the two options @schneider42 mentioned are probably the best ones, there is a third one:
* install everything(gnuradio-3.9, gr-osmosdr & gr-iridium) from source with pybombs

If you follow the commands in the the README of https://github.com/Sec42/sec-gr/ you should end up with a working gnuradio/gr-iridium build in a local prefix. Be warned that this will take around ~20G of space and likely a few hours of compile time depending on your hardware.

Sec42, 2021-08-25:
I just wrote some code which allows to use soapy instead of osmosdr. This hopefully works out of the box with gnuradio-3.9 without the need of installing gr-osmosdr.

You can check whether soapy works for you by running "SoapySDRUtil --find" with the SDR plugged in. If it can see your SDR you might be able to use it.

To try it, checkout & build the "soapy" branch, and then run:

iridium-extractor -D 4 examples/rtl-sdr-soapy.conf

If anyone tries this, please add a comment wether that works for you or not.

harrycheng20, 2021-09-10:
> https://github.com/Sec42/sec-gr/

> While the two options @schneider42 mentioned are probably the best ones, there is a third one:
>
> * install everything(gnuradio-3.9, gr-osmosdr & gr-iridium) from source with pybombs
>
> If you follow [...]

I follow the README in the Github link and it works on Ubuntu 20.04 with gnuradio-3.9! It takes around 1-hour with i7-6600U processor.

schneider42, 2021-10-01:
I've mentioned this in the readme: https://github.com/muccc/gr-iridium/blob/master/README.md#gnuradio-39-on-ubuntu-or-debian

closed 2021-10-01 by schneider42