muccc/gr-iridium #64: No signals

kevtheskin, 2021-08-06:
Hello there, I have been playing around with this with my rtlv3 dongle and patch and i get no signals. In gqrx i can see what looks like sporadic peaks at about 1.621 ghz onwards but no decode in gr-iridium. Anyone had the same issues? 

Cheers Kev

schneider42, 2021-08-06:
Can you send us a screenshot of what you see in gqrx? You are most likely to see something around 1.626 GHz

kevtheskin, 2021-08-06:
> 
> 
> Can you send us a screenshot of what you see in gqrx? You are most likely to see something around 1.626 GHz

Hi there, I am seeing what looks like intermittent wifi type peaks across the whole band. I can see inmarsat 54w ok and my location is Scotland. Thank you for the reply

kevtheskin, 2021-08-12:
Latest update . I have installed the rtlsdrblog updated driver on my windows box and I can see lots of signals using sdr#. I did the update on my linux box and i could see signals for a short while using Gqrx and Gr-iridium and now nothing. Went back to my windows box and can still see lots of signals? Confused :(

schneider42, 2021-08-12:
Which exact SDR are you using? If you don't know for sure a picture might help.

I know that R820t tuner based devices can have difficulties tuning to such high frequencies. The tend to overheat in this case and I believe the PLL for the LO loses lock in that case. Maybe sdr# somehow configures the tuner in a different way, avoiding the issue.

kevtheskin, 2021-08-12:
> Which exact SDR are you using? If you don't know for sure a picture might help.
>
> I know that R820t tuner based devices can have difficulties tuning to such high frequencies. The tend to overheat in [...]

Hi there, I have 2x Rtlsdr v3

schneider42, 2022-02-01:
Could you give the `--raw-capture` a try and upload the result somewhere? Maybe we can see something there.

schneider42, 2022-03-25:
@kevtheskin let me know if you are still working on this. Otherwise I will close the issue.