muccc/gr-iridium #66: Errors on cmake and make

BH6AZP, 2021-08-23:
There is something wrong when I compile. I try my best but still cannot solve it. Could you give me some help? Thanks.

neil@neil-virtual-machine:~/Desktop/master2/build$ cmake ..
-- Build type not specified: defaulting to release.
-- Checking for module 'mpir >= 3.0'
--   No package 'mpir' found
-- Could NOT find MPIR (missing: MPIRXX_LIBRARY MPIR_LIBRARY MPIR_INCLUDE_DIR)
-- User set python executable /usr/bin/python3
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.8.so (found suitable exact version "3.8.10") 
-- Using install prefix: /usr/local
-- Building for version: v1.0-compat-xxx-xunknown / 1.0.0git
-- No C++ unit tests... skipping
-- 
-- Checking for module SWIG
-- Found SWIG version 4.0.1.
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.8.so (found version "3.8.10") 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/neil/Desktop/master2/build
neil@neil-virtual-machine:~/Desktop/master2/build$ make
make[2]: *** No rule to make target '/usr/lib/x86_64-linux-gnu/liborc-0.4.so', needed by 'lib/libgnuradio-iridium.so.v1.0-compat-xxx-xunknown'. Stop.
make[1]: *** [CMakeFiles/Makefile2:252: lib/CMakeFiles/gnuradio-iridium.dir/all] Error 2
make: *** [Makefile:141: all] Error 2

schneider42, 2021-09-29:
Can you tell us a bit more about your setup? Which Linux distribution are you using and how did you install GNURadio?

schneider42, 2022-01-22:
@BH6AZP any update on this? Otherwise I'm going to close the issue.

closed 2022-02-01 by schneider42