muccc/gr-iridium #67: Tag releases compatible with GR 3.8 and 3.9

ryanvolz, 2021-09-20:
I suggested to @Sec42 that users would benefit from having a conda package available for `gr-iridium`. It looks like that should be easy enough dependency-wise, but it would also be really helpful for me in creating that package if you could tag some recent releases that are compatible with GNU Radio 3.8 and 3.9. Thanks!

ryanvolz, 2021-09-20:
FYI, date tags with some sort of GR branch suffix, like "YYYY.MM.DD.3.8" or "YYYY.MM.DD-maint-3.8" would be a perfectly usable version identifier that works with conda-forge.

Sec42, 2021-09-26:
I added two tags: 2021-09-26_maint-3.8 and 2021-09-26_maint-3.9. Hope these help.

3.8 will probably not get any more changes. But feel free to reach out if you need a new tag on maint-3.9

closed 2021-09-26 by Sec42

ryanvolz, 2022-01-20:
I know it's been a while, but gr-iridium is finally packaged on conda-forge as [`gnuradio-iridium`](https://github.com/conda-forge/gnuradio-iridium-feedstock). If any developers want to be added as co-maintainers, I will happily do that. I've also updated [radioconda](https://github.com/ryanvolz/radioconda), my self-contained installer, to include gr-iridium.

I don't know where support for GNU Radio 3.10 stands, but as soon as it's supported it might be helpful to make a new tag at that point. Or whenever in the future, I can make new builds of the conda package whenever you make a new tag (just let me know so I don't miss it!).

schneider42, 2022-01-20:
When making a new tag, is there a manual step needed to have radioconda pick it up (within 3.8 and 3.9)?

ryanvolz, 2022-01-20:
I don't know how exactly the magic works, but conda-forge has a mechanism for detecting new releases and automatically issuing PRs to update the package. I think that will work in this case since I think it just monitors Github's tag feed, so whenever there is a new tag I should get an updated build automatically.

As far as updating the radioconda installer, that is a manually triggered process but the build is automated. I usually kick it every month or two, or when a package like GNU Radio gets an update. People who already have radioconda or conda installed can still update packages themselves from conda-forge though, without having to use an updated installer.