muccc/gr-iridium #96: Wrong number or type 'new_time_spec_t'

AdmiralGeta, 2021-11-20:
Hi schneider, i installed Gnuradio3.9 and last gr-iridium from master branch, installation was successful but i have TypeError conflict, can you tell what I did wrong

Setup:
OS: Dragonos\Lubuntu
Receiver`s: limeSDR\RTL-SDR

[image]

schneider42, 2021-11-20:
It looks like gr-osmosdr is installed for gnuradio 3.8 (it mentions SWIG in the traceback). Your system must have a mix of GNURadio 3.9 and GNURadio 3.8. How did you install GNURadio and how did you install gr-osmosdr?

A first tip:
You must _never_ install GNURadio 3.9 from the PPA and then install gr-osmosdr bia apt. It _will_ lead to a broken installation.

schneider42, 2022-01-22:
@AdmiralGeta Any update on this? Otherwise I'm going to close this issue

AdmiralGeta, 2022-01-22:
Yes, you can close my issues, I install all instruments, thanks for your
work and help.

Best winches!

closed 2022-02-01 by schneider42