muccc/gr-iridium #98: Error: Line 316 Flow Graph

FairyballYT, 2021-12-08:
Hi, so I tried to run iridium-extractor within a VirtualBox virtual machine running DragonOS Focal Public R18 (which is based off of Lubuntu 0.14) but I kept getting this python flow graph error when trying to do an online decode from my RTL-SDR V3 dongle. I compiled/installed GNU-Radio from the 3.9.0.0 source, installed gr-iridium, and while gr-iridium will launch and find my RTL-SDR, it will crash within a few seconds due to this Line 316 Flow Graph Python error. I'm not sure how to resolve the issue, is my copy of gr-iridium missing some python files or something?
[image]

schneider42, 2021-12-08:
The name of the parameter changed from `signal_width` to `burst_width` earlier this year e13828720b2afeaaf0ff14e38e38a1005eb4b4ab

It changed on both sides though. It's almost like you got a new version of one file and an old version of another file on your system.

Can you show us the output of `find /usr -name iridium_extractor_flowgraph.py`? I'm wondering if you have this file twice.

FairyballYT, 2021-12-08:
It looks like I have the same flowgraph.py file twice. I think DragonOS came preinstalled with Python3, and then either it cane preinstalled with Python3.8 or I tried to install Python 3.8 via the terminal, sorry, foggy memory. ~~Also, I'm not sure if this file is too large, but I would also like to send a copy of some of the programs and libraries that I also tried to install via the terminal.~~
[image]

FairyballYT, 2021-12-08:
So I think this was the sequence of installs/events that I performed through the terminal:

1. I both downloaded and installed Volk 2.5 via the terminal.

2. a. I downloaded the source version of GNU Radio 3.9.0.0 from the GNU Radio Wiki, then installed that via the terminal.

2. b. I had to download the library pybind11 off the internet and install that in the terminal otherwise GNU Radio 3.9.0.0 would not install.

3. I download the latest source version of gr-iridium from from 3 days ago off Github and installed it through the terminal.

4. I tried to use the example online command to pull data from my RTL-SDR V3 dongle at 1.622Ghz, but then the flowgraph error appeared.

schneider42, 2021-12-08:
Is it possible that DragonOS came with an outdated version of gr-iridium pre-installed?

FairyballYT, 2021-12-08:
> Is it possible that DragonOS came with an outdated version of gr-iridium pre-installed?

No, DragonOS did not come pre installed with gr-iridium at all, I had to download the source code and then install gr-iridium in the terminal.

schneider42, 2021-12-08:
Can you post the output of `md5sum` for the two files installed?

FairyballYT, 2021-12-08:
> Can you post the output of `md5sum` for the two files installed?

Sure, the md5sum outputs seem to be different which could indicate that the contents of both flowgraph.py files are different. So... I now have two versions of Python installed in DragonOS? 🤔
[image]

schneider42, 2021-12-08:
This is not a question of which Python versions you have installed. The flowgraph does not come with a Python installation and the Python version does not influence the content of the flowgraph file.

You definitely have two versions of `gr-iridium` on your system though. I don't know how they got there though. Judging from the MD5 sums, one of them is from master (the one in /usr/local/lib/python3.8/dist-packages) and one is from a much earlier version (based on a commit between a7d21a68878989257d99d52481c07ee2d1ce4121..18f7d5b3fc0d1b0de14763eaeb46e15d5db7dba2 from early 2020) in /usr/lib/python3/dist-packages

Either your system came pre installed with a very dated version of gr-iridium or you did install it at some point. I suggest to delete the old version (that includes all the files in `/usr/lib/python3/dist-packages/iridium`).

FairyballYT, 2021-12-08:
> This is not a question of which Python versions you have installed. The flowgraph does not come with a Python installation and the Python version does not influence the content of the flowgraph file.
> 
> You definitely have two versions of `gr-iridium` on your system. I don't know how they got there though. Judging from the MD5 sums, one of them is from master (the one in /usr/local/lib/python3.8/dist-packages) and one is from a much earlier version (based on a commit between a7d21a68878989257d99d52481c07ee2d1ce4121..18f7d5b3fc0d1b0de14763eaeb46e15d5db7dba2 from early 2020) in /usr/lib/python3/dist-packages
> 
> Either your system came pre installed with a very dated version of gr-iridium or you did install it at some point. I suggest to delete the old version (that includes all the files in `/usr/lib/python3/dist-packages/iridium`).

Oh... I see. How would I delete an old version of a package, just by the package name "python3" in the terminal?

FairyballYT, 2021-12-08:
Using the command dpkg --list reminded me that DragonOS came preinstalled with... Python 2.7 and Python 3 and Python 3.8?

FairyballYT, 2021-12-08:
I can't find the iridium folder in the directory usr/lib/python3
[image]

FairyballYT, 2021-12-09:
Nevermind, I was searching within the /usr/lib directory, not the usr/local/lib directory
[image]
. 😅

FairyballYT, 2021-12-09:
I think when I installed the source version of gr-iridium from 3-4 days ago, it got installed to the python3/dist-packages/iridium directory. I'll leave a copy of my installed program if that helps.

Google Drive: https://drive.google.com/file/d/1Twkq0AIjTsqdH8mVjcNssqwzo01gNwqy/view?usp=sharing

FairyballYT, 2021-12-13:
Whoops, I should've said that DragonOS is based off Debian 20.04, I fixed that in the first comment.

schneider42, 2022-02-01:
Sorry for not following up here. Are you still experiencing this issue?

closed 2022-02-15 by schneider42