muccc/gr-iridium #101: iridium extractor on test data results in empty output.bits folder

YusufAtes, 2022-01-07:
$ iridium-extractor -c 1626000000 -r 2000000 -f float prbs15-2M-20dB.sigmf-data > output.bits

i use the above command and receive the output

1641554078 | i:   0/s | i_avg:   0/s | q_max:    0 | i_ok:   0% | o:    0/s | ok:   0% | ok:   0/s | ok_avg:   0% | ok:          0 | ok_avg:   0/s | d: 0
Done.

and output.bits folder is empty.

did i install gr-iridium falsely ?

schneider42, 2022-01-08:
Please give this a try: `iridium-extractor prbs15-2M-20dB.sigmf-data > output.bits`

schneider42, 2022-01-08:
Also: Are you trying this with GNURadio 3.8 or with GNURadio 3.9? I believe the `maint-3.8` branch has an issue with short recordings.

If you are using the 3.8 branch, try this:
```
cat prbs15-2M-20dB.sigmf-data prbs15-2M-20dB.sigmf-data prbs15-2M-20dB.sigmf-data prbs15-2M-20dB.sigmf-data > prbs15-2M-20dB.sigmf-dataX4
iridium-extractor -c 1626000000 -r 2000000 -f float prbs15-2M-20dB.sigmf-dataX4
```

YusufAtes, 2022-01-13:
> Also: Are you trying this with GNURadio 3.8 or with GNURadio 3.9? I believe the `maint-3.8` branch has an issue with short recordings.
>
> If you are using the 3.8 branch, try this:
>
> ```
> cat [...]

it worked thanks

closed 2022-01-13 by YusufAtes