muccc/gr-iridium #108: Receiving multiple ring alert signals from different satellites ?

YusufAtes, 2022-01-20:
I am using hackrf and maxtenna  M1621HCT-EXT IRIDIUM EXTERNAL MAGNET MOUNT ANTENNA (datasheet: https://maxtena.com/wp-content/uploads/2016/06/m1621hct-ext.pdf ) I record the iridium signals and demodulate them using iridium toolkit however for a period of time I only see IRA signals from one satellite at a time then from another satellite while in theory, I should receive signals from both satellites together periodically

[image]
The above table is from the iridium extractor output taken only the IRA signals. From the 104th satellite and 28th satellite IRA signals are missed there are none for almost 130 seconds and than 28th satellite IRA signals start and 104th satellites signals ends. We should receive 104 and 28 together at some point but we cant what is the reason for this is it not possible to receive synchronous IRA signals from different satellites in same location? 

I check the locations of the satellites from GSAT tle and check whether my location is covered by two or more satellites and i still could not receive simultaneous IRA signals.

schneider42, 2022-01-20:
Your results points to a non-optimal receiver setup: 
 - You should be able to see many more satellites and beams at any point in time
 - Your confidence values are rather low. You should see more in the 95% to 100% region
 - The gaps between ring alerts are quite large. You should see one _at least_ ever 4.32 seconds

What kind of antenna are you using? Do you think there might be RF interference where you are located?

If you share your IRA: lines, we can have a look at the SNR.

YusufAtes, 2022-01-20:
Firstly, thanks for replying so fast 

Our antenna model datasheet link is present in this link: https://maxtena.com/wp-content/uploads/2016/06/m1621hct-ext.pdf
It's a helical antenna with the following specifications
[image]

I live in an urban area where there is a high possibility for RF interference

[IRA.txt](https://github.com/muccc/gr-iridium/files/7906760/IRA.txt)
This is the txt file for all IRA lines and I paste the headers of the file in below 

IRA: i-1642621011-t1 000007105 1626256768  85%   0.001 103 DL sat:104 beam:38 pos=(+42.32/+032.30) alt=013 RAI:48 ?00 bc_sb:08 PAGE(NONE) descr_extra:0110101101011110011100100100001110001111111000

IRA: i-1642621011-t1 000054626 1626255488  84%   0.001 111 DL sat:104 beam:38 pos=(+26.83/+039.13) alt=798 RAI:48 ?00 bc_sb:08 PAGE(NONE) descr_extra:01101011010111100111001001010111110101101000001001111101100011

IRA: i-1642621011-t1 000063266 1626255232  90%   0.001 121 DL sat:104 beam:38 pos=(+26.31/+039.13) alt=801 RAI:48 ?00 bc_sb:08 PAGE(NONE) PAGE(FILL) descr_extra:001000110010001001

IRA: i-1642621011-t1 000067586 1626255232  90%   0.001 098 DL sat:104 beam:38 pos=(+38.71/+032.68) alt=015 RAI:48 ?00 bc_sb:08 PAGE(NONE) descr_extra:011010110101111001110000010000110110

IRA: i-1642621011-t1 000192824 1626320768  93%   0.001 119 DL sat:28 beam:19 pos=(+51.27/+035.79) alt=793 RAI:48 ?10 bc_sb:21 PAGE(NONE) PAGE(FILL) descr_extra:01101011000011

IRA: i-1642621011-t1 000197144 1626320640  92%   0.001 093 DL sat:28 beam:19 pos=(+37.53/+031.40) alt=014 RAI:48 ?10 bc_sb:21 PAGE(NONE) descr_extra:01101011010111100111000010

IRA: i-1642621011-t1 000210104 1626320128  91%   0.001 095 DL sat:28 beam:19 pos=(+50.24/+035.89) alt=793 RAI:48 ?10 bc_sb:21 PAGE(NONE) descr_extra:011010110101111001110011001100

IRA: i-1642621011-t1 000214424 1626320000  88%   0.001 127 DL sat:28 beam:19 pos=(+36.53/+031.52) alt=015 RAI:48 ?10 bc_sb:21 PAGE(NONE) PAGE(FILL) descr_extra:011010110101111011101000010011

IRA: i-1642621011-t1 000218743 1626319744  86%   0.001 111 DL sat:28 beam:19 pos=(+49.72/+035.92) alt=796 RAI:48 ?10 bc_sb:21 PAGE(NONE) descr_extra:01101011010111100111000011101011000011000101101101010011101000

schneider42, 2022-01-20:
Your signal level is at 0.001, which is very, very weak. Maybe it helps to increase the gain on your SDR. Is there a cable between the antenna and your SDR? Can you make that shorter?

In general active antennas are preferable. Like the "Active L-Band" antenna from https://www.rtl-sdr.com/buy-rtl-sdr-dvb-t-dongles/. Disclaimer: I'm not super happy with the performance of that antenna either, but it is better than a passive antenna with 1.5 meters of cable.

YusufAtes, 2022-01-22:
Thanks, I will apply your recommendations also, for the example hackrf config files IF and BB gain are 40 and 20. Is there any documentation on how to choose IF and BB gain because there are no explanations available for the optimal receiver setup.

YusufAtes, 2022-01-23:
Also, I noticed that in the readme signal detection threshold is 18 dB but in the output using verbose, it says that the signal detection threshold is 8.5 dB is this an error in the readme?

schneider42, 2022-02-01:
I think there were some changes to how the detection threshold is handled. At some point the debug output might not have reflected the true behaviour. Do you know which version of the gr-iridium you are using? How did you install it?

The gain values for the hackrf were chosen experimentally by looking at the watefall it produced. That is so long ago that I can't really comment on that anymore. If these are the gains you are using, you should expect quite a bit higher signal levels though. I think it would be helpful if you could share a recording (using the `--raw-capture` option), if you are willing to do so. You might need a more recent gr-iridium version though if the option is not available in your installation.

amitpathania, 2022-03-09:
> I am using hackrf and maxtenna M1621HCT-EXT IRIDIUM EXTERNAL MAGNET MOUNT ANTENNA (datasheet: https://maxtena.com/wp-content/uploads/2016/06/m1621hct-ext.pdf ) I record the iridium signals and [...]

how are you getting this output in tabulated format? can you share the code get output in this format??

YusufAtes, 2022-03-09:
> how are you getting this output in tabulated format? can you share the code get output in this format??

I don't get the output with this format I run extra python code after saving the file. I deleted the code as I remember it was a simple python code using pandas and dfi.export

closed 2022-03-09 by YusufAtes