muccc/gr-iridium #118: TypeError: Wrong number or type of arguments for overloaded function 'new_time_spec_t'

dukuilanu, 2022-02-20:
Hi there,

I get the following error when trying to run iridium-extractor with an rtl-sdr blog v3. Any help would be much appreciated:

raspberrypi% iridium-extractor  ./rtl-sdr.conf > ~/output.bits
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.2.0
built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
[INFO] [UHD] linux; GNU C++ version 10.2.0; Boost_107100; UHD_3.15.0.0-4
Found Rafael Micro R820T tuner
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 2000000.052982 Hz
[R82XX] PLL not locked!
Traceback (most recent call last):
  File "/usr/local/bin/iridium-extractor", line 342, in <module>
    tb = iridium.iridium_extractor_flowgraph.FlowGraph(center_frequency=center, sample_rate=sample_rate, decimation=decimation, 
  File "/usr/local/lib/python3.9/dist-packages/iridium/iridium_extractor_flowgraph.py", line 137, in __init__
    source.set_time_now(osmosdr.time_spec_t(int(t0), t0 - int(t0)))
  File "/usr/lib/python3/dist-packages/osmosdr/osmosdr_swig.py", line 936, in __init__
    _osmosdr_swig.time_spec_t_swiginit(self, _osmosdr_swig.new_time_spec_t(*args))
TypeError: Wrong number or type of arguments for overloaded function 'new_time_spec_t'.
  Possible C/C++ prototypes are:
    osmosdr::time_spec_t::time_spec_t(double)
    osmosdr::time_spec_t::time_spec_t(time_t,double)
    osmosdr::time_spec_t::time_spec_t(time_t,long,double)

versions of stuff:

gcc (Raspbian 10.2.1-6+rpi1) 10.2.1 20210110

gnuradio 3.8

raspberrypi% git branch
* maint-3.8

raspberry os bullseye (which is fairly new)

schneider42, 2022-02-20:
I can reproduce this on 3.8. CI did not help here as without unit tests this needs an actual SDR connected.

schneider42, 2022-02-20:
Waiting for the backport

schneider42, 2022-02-20:
@dukuilanu can you give main-3.8 another try? should be fixed now.

dukuilanu, 2022-02-20:
Confirmed working. Thank you kindly.

PR #119 (2022-02-20): iridium_extractor: use a double to init time estimate
Using an int and a double caused gnuradio 3.8 to fail at runtime.

Closes #118
commit 9db591d4 (2022-02-20): iridium_extractor: use a double to init time estimate

Using an int and a double caused gnuradio 3.8 to fail at runtime.

Closes #118
commit 551c998b (2022-02-20): iridium_extractor: use a double to init time estimate

Using an int and a double caused gnuradio 3.8 to fail at runtime.

Closes #118
closed 2022-02-20 by dukuilanu