muccc/gr-iridium #121: problem using cmake during installation

michael040101, 2022-03-04:
Hi,
Im trying to install the program on ubunto 20.04 with gnuradio 3.9, and this is what happens during the cmake stage,

_michael@ubuntu:~/Desktop/gr-iridium/build$ cmake ..
-- Build type not specified: defaulting to release.
-- Using GMP.
-- User set python executable /usr/bin/python3
-- Checking for module 'sndfile'
--   No package 'sndfile' found
-- Could NOT find SNDFILE (missing: SNDFILE_LIBRARIES SNDFILE_INCLUDE_DIRS) 
-- Extracting version information from git describe...
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
-- Using install prefix: /usr/local
-- Building for version: 2022-02-02_maint-3.9-3-g9db591d4 / 1.0.0git
-- No C++ unit tests... skipping
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
-- PYTHON and GRC components are enabled
-- CMAKE_INSTALL_PREFIX: /usr/local
-- GR_PYTHON_DIR: lib/python3.8/dist-packages
-- GR_PYTHON_DIR: lib/python3.8/dist-packages
-- Python checking for pygccxml - found
-- Configuring done
CMake Error at /usr/lib/cmake/pybind11/pybind11Tools.cmake:131 (add_library):
  Target "iridium_python" links to target "sndfile::sndfile" but the target
  was not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?
Call Stack (most recent call first):
  /usr/lib/x86_64-linux-gnu/cmake/gnuradio/GrPybind.cmake:241 (pybind11_add_module)
  python/bindings/CMakeLists.txt:42 (GR_PYBIND_MAKE_OOT)

CMake Error at /usr/lib/cmake/pybind11/pybind11Tools.cmake:131 (add_library):
  Target "iridium_python" links to target "sndfile::sndfile" but the target
  was not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?
Call Stack (most recent call first):
  /usr/lib/x86_64-linux-gnu/cmake/gnuradio/GrPybind.cmake:241 (pybind11_add_module)
  python/bindings/CMakeLists.txt:42 (GR_PYBIND_MAKE_OOT)

CMake Error at lib/CMakeLists.txt:32 (add_library):
  Target "gnuradio-iridium" links to target "sndfile::sndfile" but the target
  was not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?

CMake Error at lib/CMakeLists.txt:32 (add_library):
  Target "gnuradio-iridium" links to target "sndfile::sndfile" but the target
  was not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?

-- Generating done
CMake Generate step failed.  Build files cannot be regenerated correctly._

Do you have any idea how can i solve this issue?

thanks in advance!

schneider42, 2022-03-04:
You are missing the `libsndfile1-dev` package. You can install it like this: `sudo apt install libsndfile1-dev`

closed 2022-03-04 by schneider42

michael040101, 2022-03-04:
Thanks a lot!

now im having issue with the make stage,

michael@ubuntu:~/Desktop/gr-iridium-maint-3.8/build$ make
Scanning dependencies of target gnuradio-iridium
[  6%] Building CXX object lib/CMakeFiles/gnuradio-iridium.dir/fft_burst_tagger_impl.cc.o
[ 12%] Building CXX object lib/CMakeFiles/gnuradio-iridium.dir/iuchar_to_complex_impl.cc.o
[ 18%] Building CXX object lib/CMakeFiles/gnuradio-iridium.dir/tagged_burst_to_pdu_impl.cc.o
[ 25%] Building CXX object lib/CMakeFiles/gnuradio-iridium.dir/burst_downmix_impl.cc.o
[ 31%] Building CXX object lib/CMakeFiles/gnuradio-iridium.dir/pdu_null_sink_impl.cc.o
[ 37%] Building CXX object lib/CMakeFiles/gnuradio-iridium.dir/iridium_qpsk_demod_cpp_impl.cc.o
[ 43%] Building CXX object lib/CMakeFiles/gnuradio-iridium.dir/pdu_round_robin_impl.cc.o
make[2]: *** No rule to make target '/usr/lib/x86_64-linux-gnu/liborc-0.4.so', needed by 'lib/libgnuradio-iridium.so.v1.0-compat-xxx-xunknown'.  Stop.
make[1]: *** [CMakeFiles/Makefile2:252: lib/CMakeFiles/gnuradio-iridium.dir/all] Error 2
make: *** [Makefile:141: all] Error 2

note that im usung now gnuradio3.8

any idea on how to fix that?