muccc/gr-iridium #122: errors during make stage

michael040101, 2022-03-04:
Hi,

Im using gnuradio3.9 on ubuntu 20.04.
Im having issue with the make stage, 

michael@ubuntu:~/Desktop/gr-iridium/build$ make
Scanning dependencies of target gnuradio-iridium
[  4%] Building CXX object lib/CMakeFiles/gnuradio-iridium.dir/fft_burst_tagger_impl.cc.o
[  8%] Building CXX object lib/CMakeFiles/gnuradio-iridium.dir/iuchar_to_complex_impl.cc.o
[ 12%] Building CXX object lib/CMakeFiles/gnuradio-iridium.dir/tagged_burst_to_pdu_impl.cc.o
[ 16%] Building CXX object lib/CMakeFiles/gnuradio-iridium.dir/burst_downmix_impl.cc.o
[ 20%] Building CXX object lib/CMakeFiles/gnuradio-iridium.dir/pdu_null_sink_impl.cc.o
[ 24%] Building CXX object lib/CMakeFiles/gnuradio-iridium.dir/iridium_qpsk_demod_impl.cc.o
[ 28%] Building CXX object lib/CMakeFiles/gnuradio-iridium.dir/pdu_round_robin_impl.cc.o
[ 32%] Building CXX object lib/CMakeFiles/gnuradio-iridium.dir/frame_sorter_impl.cc.o
[ 36%] Building CXX object lib/CMakeFiles/gnuradio-iridium.dir/iridium_frame_printer_impl.cc.o
make[2]: *** No rule to make target '/usr/lib/x86_64-linux-gnu/liborc-0.4.so', needed by 'lib/libgnuradio-iridium.so.2022-02-02_maint-3.9-3-g9db591d4'.  Stop.
make[1]: *** [CMakeFiles/Makefile2:249: lib/CMakeFiles/gnuradio-iridium.dir/all] Error 2
make: *** [Makefile:141: all] Error 2

Do you have any idea how to fix it?

thanks in advance

schneider42, 2022-03-05:
Try installing `liborc-0.4-dev`

michael040101, 2022-03-06:
It works
Thank you!

closed 2022-03-06 by schneider42