muccc/gr-iridium #125: cannot import name 'soapy' from 'gnuradio' when used with gnuradio 3.8

alphafox02, 2022-03-10:
I'm wondering if the gr-soapy OOT module for GR 3.8 is taken into account by gr-iridium or if the soapy-source conf files are only for the newer GR versions with the in-tree soapy as mentioned here?

https://wiki.gnuradio.org/index.php/Soapy

DragonOS has this version of gr-soapy for 3.8 built in,

https://gitlab.com/librespacefoundation/gr-soapy/-/tree/master/

However, when trying to use soapy-source instead of osmosdr-source at the top of conf files, I get the following,

```
iridium-extractor -D 4 /usr/src/gr-iridium/examples/rsp1a.conf > output.bits
Traceback (most recent call last):
  File "/usr/local/bin/iridium-extractor", line 356, in <module>
    tb = iridium.iridium_extractor_flowgraph.FlowGraph(center_frequency=center, sample_rate=sample_rate, decimation=decimation, 
  File "/usr/local/lib/python3.8/dist-packages/iridium/iridium_extractor_flowgraph.py", line 196, in __init__
    from gnuradio import soapy
ImportError: cannot import name 'soapy' from 'gnuradio' (/usr/lib/python3/dist-packages/gnuradio/__init__.py)
```

alphafox02, 2022-03-10:
By changing line 196 
from gnuradio import soapy 
to 
import soapy
I seemed to have moved further, and I could be wrong, but I suspect "type" is in reference to device type as shown within gnu radio companion when looking at the soapy source / sink options.

```
iridium-extractor -D 4 /usr/src/gr-iridium/examples/rsp1a.conf > output.bits
Traceback (most recent call last):
  File "/usr/local/bin/iridium-extractor", line 356, in <module>
    tb = iridium.iridium_extractor_flowgraph.FlowGraph(center_frequency=center, sample_rate=sample_rate, decimation=decimation, 
  File "/usr/local/lib/python3.8/dist-packages/iridium/iridium_extractor_flowgraph.py", line 206, in __init__
    source = soapy.source(dev, "fc32", 1, '', stream_args, tune_args, settings)
TypeError: make() missing 1 required positional argument: 'type'
```

schneider42, 2022-03-10:
API might be different. We certainly only tested with 3.9/3.10. We'll try to have a look. Please don't defer looking into this though, I think you are close to a solution.

alphafox02, 2022-03-10:
I’ll keep trying. I was sitting here wondering.. maybe because DragonOS has python2 and 3 with 2 being the default that maybe I needed to tweak headers to include python3. But then that doesn’t seem to be the case since I can run iridium extractor without specifying python3. I’ll keep messing around. 

What I’m doing is building a 3.8 flow graph manually with the gr-soapy source and then looking at the generated py file for clues.

schneider42, 2022-03-10:
`iridium-extractor` explicitly launches Python3, you don't have to specify it:

https://github.com/muccc/gr-iridium/blob/master/apps/iridium-extractor#L1

alphafox02, 2022-03-11:
I can confirm that the attached modifications to the iridium extractor flow graph works on Gnuradio 3.8 with the OOT gr-soapy module. The included conf file for the SDRPlay RSP1A is currently the only conf and device I've tested myself on DragonOS.

I hope that someone can review, create a proper pull request, and merge as seen fit. I'm not the author of the changes, only the middle person that's confirmed the changes work. @viperbjk receives the credit.

[rsp1a.zip](https://github.com/muccc/gr-iridium/files/8230800/rsp1a.zip)
[iridium_extractor_flowgraph.zip](https://github.com/muccc/gr-iridium/files/8230802/iridium_extractor_flowgraph.zip)

alphafox02, 2022-03-12:
Just a note so I don't forget, the lines dealing with printing the bandwidth were commented out, I believe do to the get/set bandwidth for soapy not working correctly, so this does effect osmosdr and soapy with both of them not showing to the user what the bandwidth is actually set to when it first starts. Just something to point out.

closed 2022-06-06 by alphafox02