muccc/gr-iridium #126: weird issue when recording data

michael040101, 2022-03-10:
Hi,
Im using ubuntu 20.04 and gnuradio 3.8.5, and i get the following error,

michael@ubuntu:~/Desktop/gr-iridium-maint-3.8$ iridium-extractor -D 4 examples/rtl-sdr.conf > output.bits
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.1.0
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
[INFO] [UHD] linux; GNU C++ version 9.2.1 20200304; Boost_107100; UHD_3.15.0.0-2build5
Found Rafael Micro R820T tuner
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 2000000.052982 Hz
[R82XX] PLL not locked!
(RF) Gain: 19.7 (Requested 19)
WARNING: Gain if not supported by source!
Supported gains: ('LNA',)
Warning: Setting antenna to RX
1646954261 | i:   0/s | i_avg:   0/s | q_max:    0 | i_ok:   0% | o:    0/s | ok:   0% | ok:   0/s | ok_avg:   0% | ok:          0 | ok_avg:   0/s | d: 0
Traceback (most recent call last):
  File "/usr/local/bin/iridium-extractor", line 373, in <module>
    tb.run()
  File "/usr/local/lib/python3.8/dist-packages/iridium/iridium_extractor_flowgraph.py", line 435, in run
    self.start(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/gnuradio/gr/top_block.py", line 111, in start
    top_block_start_unlocked(self._impl, max_noutput_items)
  File "/usr/lib/python3/dist-packages/gnuradio/gr/runtime_swig.py", line 4832, in top_block_start_unlocked
    return _runtime_swig.top_block_start_unlocked(r, max_noutput_items)
RuntimeError: list contains invalid format!

Do you have any idea why does it happen?

thanks!

RobertSwirsky, 2024-09-17:
I've never been able to solve this either. I've just given up with the rtl-sdr and this software. The rtl-sdr works fine with other GnuRadio-based programs. 

I tried 3.8-maint, 3.9-maint, and 3.10-maint as well as "main" branches of GnuRadio, but none of them will work with my rtl-sdr.

I have other SDRs here, and the work with iridium-extractor. It's just the rtl-sdr. Given the popularity of the rtl-sdr, I'm surprised nobody's fixed this.

Sec42, 2024-09-25:
@RobertSwirsky 
I just tried it, and gr-iridium can talk to rtl-sdr just fine. You can use either osmosdr or soapy to talk to it, depending on what you installed.

For soapy, you can test by running "SoapySDRUtil --find" to see if it picks up your rtl-sdr. For gr-osmosdr you can run "osmocom_fft" and see if it picks it up. If either of these work, iridium-extractor should also work -- the corresponding config files are "examples/rtl-sdr.conf" for osmosdr and "examples/rtl-sdr-soapy.conf" for soapy.

That all said: from my own experiments nearly all rtl-sdrs I tested are quite bad at receiving at ~1.626GHz and thus are not really suited for iridium reception.