muccc/gr-iridium #138: Uisng bb60 signalhound sdr

amitpathania, 2022-03-29:
Can somebody help me how one can use **bb60c signalhound sdr** with gr-iridium instead of RTL or HACKRF SDR? Manual of sdr can be found at https://signalhound.com/sigdownloads/BB60C/BB60C-User-Manual.pdf

schneider42, 2022-03-30:
You could install https://github.com/SignalHound/soapy-bb60 and try to use it using a Soapy source as outlined here: https://github.com/muccc/gr-iridium#soapy-source-section

First step would be to install the software mentioned above and post the output of `SoapySDRUtil --info`.

alphafox02, 2022-03-31:
I was just testing the bb60c on DragonOS (20.04) and found it to work with SoapySDR and GQRX, SDR++ via Soapy, etc.. however I only had it for a day and didn’t get to throughly test with GR. The directions above should work.

amitpathania, 2022-04-18:
The output of SoapySDRUtil --info:

`##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.7.1-myriadrf1~ubuntu18.04
API Version: v0.7.1
ABI Version: v0.7
Install root: /usr
Search path:  /usr/lib/x86_64-linux-gnu/SoapySDR/modules0.7
Search path:  /usr/local/lib/x86_64-linux-gnu/SoapySDR/modules0.7                (missing)
Search path:  /usr/local/lib/SoapySDR/modules0.7                                 (missing)
Module found: /usr/lib/x86_64-linux-gnu/SoapySDR/modules0.7/libHackRFSupport.so  (0.3.3)
Module found: /usr/lib/x86_64-linux-gnu/SoapySDR/modules0.7/libairspySupport.so  (0.1.2)
Module found: /usr/lib/x86_64-linux-gnu/SoapySDR/modules0.7/libbladeRFSupport.so (0.4.1)
Module found: /usr/lib/x86_64-linux-gnu/SoapySDR/modules0.7/libmiriSupport.so    (0.2.6)
Module found: /usr/lib/x86_64-linux-gnu/SoapySDR/modules0.7/libosmosdrSupport.so (0.2.6)
Module found: /usr/lib/x86_64-linux-gnu/SoapySDR/modules0.7/libremoteSupport.so  (0.5.1)
Module found: /usr/lib/x86_64-linux-gnu/SoapySDR/modules0.7/librfspaceSupport.so (0.2.6)
Module found: /usr/lib/x86_64-linux-gnu/SoapySDR/modules0.7/librtlsdrSupport.so  (0.3.0)
Module found: /usr/lib/x86_64-linux-gnu/SoapySDR/modules0.7/libuhdSupport.solinux; GNU C++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown

     (0.3.5)
Available factories... airspy, bladerf, hackrf, miri, osmosdr, remote, rfspace, rtlsdr, uhd
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]
`

alphafox02, 2022-04-18:
Download DragonOS Focal here,

https://sourceforge.net/projects/dragonos-focal/

Grab this zip here, unzip and sudo cp the contents to /usr/local/lib
https://drive.google.com/file/d/1LgG2amaWEDduA9A_NANNhNKB7AcVeElc/view?usp=sharing

Grab the sh_usb.rules from here and place in /etc/udev/rules.d/
https://drive.google.com/file/d/1Wu103Rg4FDbEmYeXxK7zvz_rEI1dIuWj/view?usp=sharing

The follow the BB60C Soapy install steps 
https://github.com/SignalHound/soapy-bb60#installation

This should get you support (but I've also only tested it on my install), gr-iridium is also built in already located in /usr/src/

This is part of my output from SoapySDRUtil. It's possible the files above will work on your 18.04 install. 

```
Module found: /usr/local/lib/SoapySDR/modules0.7/libPlutoSDRSupport.so           (0.2.0-472e9a0)
Module found: /usr/local/lib/SoapySDR/modules0.7/libXTRXSupport.so
Module found: /usr/local/lib/SoapySDR/modules0.7/libairspyhfSupport.so           (0.1.1)
Module found: /usr/local/lib/SoapySDR/modules0.7/libbb60Support.so
Module found: /usr/local/lib/SoapySDR/modules0.7/libsdrPlaySupport.so            (0.3.0-208f95d)
Available factories... airspy, airspyhf, bb60c, bladerf, hackrf, lime, miri, osmosdr, plutosdr, redpitaya, remote, rtlsdr, sdrplay, uhd, xtrx
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]
```

amitpathania, 2022-04-30:
Issue still not resolved. Getting following error while running this `cmake -D CMAKE_FIND_DEBUG_MODE=ON ..
`

`CMake Error: The source directory "/home/lenevo/Documents" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.
`

schneider42, 2022-04-30:
Looks like the BB60's Soapy driver is based on an outdated version of SoapySDR (0.4 is at least 6 years old). I think you need to talk to Signalhound to update their driver.

The other option for you might be to record raw IQ data to disk and process it offline. I don't know the tools for the BB60, but I assume the offer something like that. gr-iridium is able to process most formats.

amitpathania, 2022-05-10:
$ SoapySDRUtil -info
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.8.1-g9cbaa3cf
API Version: v0.8.0
ABI Version: v0.8
Install root: /usr/local
Search path:  /usr/local/lib/SoapySDR/modules0.8
Module found: /usr/local/lib/SoapySDR/modules0.8/libbb60Support.so
Available factories... bb60c
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]

schneider42, 2022-05-14:
Looks like you managed to get support. What happens if you try this:

```
[soapy-source]
driver=bb60

sample_rate=10000000
center_freq=1622000000
bandwidth=9000000

# Might need to be adjusted
gain=40
```