muccc/gr-iridium #149: Error while using bladerf SDR

amitpathania, 2022-05-03:
i am using gr_iridium extractor with Bladerf SDR. I am using default bladerf conf file but I am not getting any bursts.
i am using command `  iridium-extractor -D 4 examples/bladerf-10msps.conf`  and getting following output:-
`gr-osmosdr v0.1.4-127-g4d83c606 (0.1.5git) gnuradio 3.7.13.4
built-in source types: file osmosdr fcd rtl rtl_tcp miri hackrf bladerf rfspace airspy soapy redpitaya freesrp 
[bladeRF common] init: DEBUG: entering initialization
[bladeRF source] init: Opening Nuand bladeRF with device identifier string '*:instance=0'
[INFO @ host/libraries/libbladeRF/src/helpers/version.c:82] Firmware version (v2.4.0) is newer than entries in libbladeRF's compatibility table. Please update libbladeRF if problems arise.
[INFO @ host/libraries/libbladeRF/src/helpers/version.c:106] FPGA version (v0.11.0) is newer than entries in libbladeRF's compatibility table. Please update libbladeRF if problems arise.
[WARNING @ host/libraries/libbladeRF/src/board/bladerf1/bladerf1.c:1705] RX DC calibration table not found. Manual gain control will be used instead.
[INFO @ host/libraries/libbladeRF/src/board/bladerf1/bladerf1.c:1706] To enable AGC, see "Generating a DC offset table" at https://github.com/Nuand/bladeRF/wiki/DC-offset-and-IQ-Imbalance-Correction
[bladeRF source] Device: Nuand bladeRF Serial # eb62...dcb8 FW v2.4.0 FPGA v0.11.0
[bladeRF source] init: Buffers: 512, samples per buffer: 4096, active transfers: 32
[bladeRF source] bladerf_source_c: DEBUG: initialization complete
(RF) Gain: 6.0 (Requested 6)
FOUND lna
FOUND rxvga1
FOUND rxvga2
WARNING: Gain vga2 not supported by source!
Supported gains:FOUND lna
FOUND rxvga1
FOUND rxvga2
 ('system', 'lna', 'rxvga1', 'rxvga2')
FOUND lna
FOUND rxvga1
FOUND rxvga2
WARNING: Gain vga1 not supported by source!
Supported gains:FOUND lna
FOUND rxvga1
FOUND rxvga2
 ('system', 'lna', 'rxvga1', 'rxvga2')
Bandwidth: 10000000.0 (Requested 9000000)
Warning: Setting antenna to RX1
1651576029 | i:   0/s | i_avg:   0/s | q:    0 | q_max:    0 | o:   0/s | ok:   0% | ok:   0/s | ok_avg:   0% | ok:          0 | ok_avg:   0/s | d: 0
[bladeRF source] start: DEBUG: starting source
1651576030 | i:   0/s | i_avg:   0/s | q:    0 | q_max:    0 | o:   0/s | ok:   0% | ok:   0/s | ok_avg:   0% | ok:          0 | ok_avg:   0/s | d: 0

`

schneider42, 2022-05-14:
Please do not use GNU Radio 3.7 anymore. It is obsolete and there is no further development for this version happening anymore.

Please make sure that your libbladerf version has support for the firmware and FPGA images.

I've pushed a change to the master branch which makes use of the system gain of BladeRF devices. Please either update gr-iridium or apply this patch:

```diff
diff --git a/examples/bladerf-10msps.conf b/examples/bladerf-10msps.conf
index 418e1bf..ab555ec 100644
--- a/examples/bladerf-10msps.conf
+++ b/examples/bladerf-10msps.conf
@@ -3,7 +3,7 @@ sample_rate=10000000
 center_freq=1622000000
 bandwidth=9000000
 
-gain=6
-vga1_gain=30
-vga2_gain=12
+# Set the system gain to 40 dB
+# Maximum is 60 dB according to libbladerf
+gain=40
 
```

closed 2022-05-14 by schneider42

amitpathania, 2022-06-01:
I have tried using new conf file for BLADE RF SDR. There is no error being displayed about gain now. However, I am still getting 0 signal values and no data. When I am using HACKRF SDR with same antenna at same location, I am getting  data/ iridium bursts. 
Can you help me to get software working wit BLADERF SDR?