muccc/gr-iridium #154: how to log the IQ samples of all the valid IRIDIUM packets

zhusiyu6, 2022-05-23:
In paper "PAST-AI: Physical-layer Authentication of Satellite Transmitters via Deep Learning",author modified the gr-iridium model to log the IQ samples of all the valid IRIDIUM packets. But I found that these modules are closed. So is there any way to modify them?

Sec42, 2022-05-23:
We are not the authors of said paper, so I can't tell you anything about their code.

`iridium-extractor` specifically has the `--debug-id` command line option to log iq values of a specific signal. This will write the `iq` values at different stages of the reception process via the `write_data_c` function
https://github.com/muccc/gr-iridium/blob/8ebf2a81543497f6995cd63ec29e1b22fc8a73a7/lib/burst_downmix_impl.cc#L42

I think that would be a good starting point if you want to replicate their work.

zhusiyu6, 2022-05-24:
I  want to ask what is the specific meaning of the parameter value after - debug-id

zhusiyu6, 2022-05-24:
> We are not the authors of said paper, so I can't tell you anything about their code.
>
> `iridium-extractor` specifically has the `--debug-id` command line option to log iq values of a specific signal. [...]

I want to ask what is the specific meaning of the parameter value after - debug-id

Sec42, 2022-05-24:
This parameter is documented in the readme.