muccc/gr-iridium #156: SDR … loosing samples ..

omon316, 2022-07-08:
After collectring data with iridium-extractor I receive „Warning: your SDR semms to be loosing samples. ~2000k smples lost (100%)
Done.“

the output.bits file remains empty

Sec42, 2022-07-09:
I don't understand how you can say that there was no warning or error when you start this issue with a waning message?

The message means that the code did not receive any samples/data from your SDR. Assuming you were using  2 Msps/s, 2000k samples corresponds to one second of listening time. If this message appears only at the beginning of a longer run, it may just be you SDR being slow to start up.

iridium-extractor outputs statistics to stderr every second while running. These numbers would give more insight on what is going on.

omon316, 2022-07-09:
according to ceaxecuter on youtube 

on the actual DragonOS release:
"The new way of running the extractor does not use the “A:OK”, instead it looks like this as an example
iridium-extractor -D 4 examples/hackrf.conf > output.bits" 

Then it started to write data into the output file. 

But when I parsed it and use ./stats-voc.py /.../output2.parsed      I'll get an error 

42
button=1, .....
button=3, ...
Error ratio: -nan
A: 0.0(unkown) of 0.0 (unknown) ??,?%