muccc/gr-iridium #159: No supported devices found

oldirty91, 2022-08-30:
Getting the following error bellow when attempting to run extractor with RTL-SDR. i have un-commented the 'device_args' line in the examples/rtl-sdr file. Not sure where else to look to troubleshoot?? Thanks.

Command is:
$ iridium-extractor -D 4 examples/rtl-sdr.conf > output.bits

Error:
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.10.1.1
built-in source types: file rtl_tcp rfspace redpitaya 
Traceback (most recent call last):
  File "/usr/local/bin/iridium-extractor", line 365, in <module>
    tb = iridium.iridium_extractor_flowgraph.FlowGraph(center_frequency=center, sample_rate=sample_rate, decimation=decimation, 
  File "/usr/local/lib/python3.10/dist-packages/iridium/iridium_extractor_flowgraph.py", line 151, in __init__
    source = osmosdr.source(args=d['device_args'])
RuntimeError: No supported devices found (check the connection and/or udev rules).

schneider42, 2022-08-30:
Do you have other SDR applications installed which are able to access the RTL-SDR? Maybe this is a permission issue.

oldirty91, 2022-08-30:
Yes I have rtl-sdr package installed as well as cubicSdr. Neither of those
are running when I attempt to run gr-iridium.

Worth removing those? Perhaps adding a second sdr and specifying 1 instead
of 0 might help if it’s a permission issue?

oldirty91, 2022-08-31:
Update:

I switched to using the 'rtl-sdr-soapy.conf' configuration and I'm able to now read the SDR. However I have hit a wall with the sample rate and decimation.

1. I get the following error when attempting the 2000000 rate:

ValueError: source: Unsupported sample rate (2000000.000000).  Rate must be in the range 250000.000000, 1024000.000000, 1536000.000000, 1792000.000000, 1920000.000000, 2048000.000000, 2160000.000000, 2560000.000000, 2880000.000000, 3200000.000000

When I attempt to use any of the values provided there I get an error that sample rate must be divisible by 100000. So that leaves 3200000.

2. When I run the 3200000 rate I get:

RuntimeError: Selected sample rate and decimation can not be matched. Please try a different combination. Sample rate divided by decimation must be a multiple of 250000.

So obviously 3200000 will not work, so I attempt to run without -D and turn decimation off, however I get the same error.

Any ideas on how to get this running? How do I run without decimation? Is there a sample rate that will work that I'm overlooking?

Thank You.

schneider42, 2022-08-31:
Sadly soapy has this limitation in the versions available in current distributions. See https://github.com/pothosware/SoapyRTLSDR/issues/59

As a workaround you could try to install a more recent version of SoapyRTLSDR from source: https://github.com/pothosware/SoapyRTLSDR/releases/tag/soapy-rtl-sdr-0.3.3

I might dig out some code which adds a resampler at the input of gr-iridium, but that might take a while.

oldirty91, 2022-08-31:
Ok, thanks for the quick reply!

I think I'll return to trying to get the osmosdr-source working. Seems to be that there others who have gotten the rtl-sdr.conf to work with the sdr I'm using so I must be missing something. I have completely removed gr-iridium and rebuilt the package and still getting the same issue. I attempted to run osmosdr fft commands  and received similar issues. Seems that osmosdr is the problem.