muccc/gr-iridium #164: Issue with boost format

gariac, 2022-09-10:
`
 cmake -B build     
-- Build type not specified: defaulting to release.
-- Checking for module 'gmp'
--   No package 'gmp' found
-- Using GMP.
-- Extracting version information from git describe...
-- Found pybind11: /usr/include (found version "2.9.2")
-- Using install prefix: /usr/local
-- Building for version: v1.0.0-28-gf90bc7d7 / 1.0.0git
-- No C++ unit tests... skipping
-- User set python executable /usr/bin/python3.9
-- PYTHON and GRC components are disabled
-- Configuring done
-- Generating done
-- Build files have been written to: /usr/local/src/gr-iridium/build
linux-yxjo:/usr/local/src/gr-iridium # cmake --build build
Consolidate compiler generated dependencies of target gnuradio-iridium
[  7%] Building CXX object lib/CMakeFiles/gnuradio-iridium.dir/iridium_frame_printer_impl.cc.o
/usr/local/src/gr-iridium/lib/iridium_frame_printer_impl.cc:13:14: error: ‘format’ has not been declared in ‘boost’
   13 | using boost::format;
      |              ^~~~~~
/usr/local/src/gr-iridium/lib/iridium_frame_printer_impl.cc: In member function ‘void gr::iridium::iridium_frame_printer_impl::handler(const pmt_t&)’:
/usr/local/src/gr-iridium/lib/iridium_frame_printer_impl.cc:85:18: error: ‘format’ was not declared in this scope; did you mean ‘fmt::v7::format’?
   85 |     std::cout << format("%012.4f ") % ((timestamp - d_t0) / 1000000.);
      |                  ^~~~~~
      |                  fmt::v7::format
In file included from /usr/include/spdlog/fmt/fmt.h:25,
                 from /usr/include/spdlog/common.h:36,
                 from /usr/local/include/gnuradio/logger.h:36,
                 from /usr/local/include/gnuradio/custom_lock.h:15,
                 from /usr/local/include/gnuradio/buffer.h:15,
                 from /usr/local/include/gnuradio/buffer_double_mapped.h:15,
                 from /usr/local/include/gnuradio/io_signature.h:23,
                 from /usr/local/include/gnuradio/basic_block.h:15,
                 from /usr/local/include/gnuradio/block.h:17,
                 from /usr/local/src/gr-iridium/lib/../include/iridium/iridium_frame_printer.h:11,
                 from /usr/local/src/gr-iridium/lib/iridium_frame_printer_impl.h:11,
                 from /usr/local/src/gr-iridium/lib/iridium_frame_printer_impl.cc:8:
/usr/include/fmt/core.h:2074:36: note: ‘fmt::v7::format’ declared here
 2074 | FMT_INLINE std::basic_string<Char> format(const S& format_str, Args&&... args) {
      |                                    ^~~~~~
gmake[2]: *** [lib/CMakeFiles/gnuradio-iridium.dir/build.make:188: lib/CMakeFiles/gnuradio-iridium.dir/iridium_frame_printer_impl.cc.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:212: lib/CMakeFiles/gnuradio-iridium.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
`

system info:

gnuradio-config-info -v
v3.11.0.0git-226-g9c64b6df

Linux linux-yxjo 5.3.18-150300.59.90-default #1 SMP Tue Aug 9 08:49:53 UTC 2022 (127973b) x86_64 x86_64 x86_64 GNU/Linux

schneider42, 2022-09-11:
Hi, can you tell me which Linux distribution you are using?

schneider42, 2022-09-11:
Please try the current master branch, might be fixed

gariac, 2022-09-11:
The new code compiles and installs. Thanks.

If you still care I am using opensuse 15.3. I swear I am going to do Ubuntu on the next notebook. I looks like Debian based systems these days are better supported. But at least you know it compiles on Opensuse and I assume whatever rpm based distributions exist.

closed 2022-09-11 by gariac