muccc/gr-iridium #173: Can't seem to control frequency of capture... bladeRF, hackRF, and rtl-sdr

jamesmac10, 2023-03-19:
Hi there, 
First things first thank you so much for all your efforts developing this suite of tools. Extremely impressive. The only issue I seem to be having, everything is perfect except it seems no matter what center_freq I specify ( I can see a lot of activity here on 1625000000 so I'm trying to set iridium-extractor to receive this area of the spectrum) the frequency that iridium-extractor actually starts at is way off. What am I doing wrong please, or possibly what is not working? Here is the output with the bladeRF (I also have same results with hackrf and the rtl sdr dongle).
ibusb: warning [libusb_exit] device 4.7 still referenced
libusb: warning [libusb_exit] device 4.6 still referenced
libusb: warning [libusb_exit] device 4.4 still referenced
libusb: warning [libusb_exit] device 4.5 still referenced
libusb: warning [libusb_exit] device 4.3 still referenced
libusb: warning [libusb_exit] device 4.2 still referenced
libusb: warning [libusb_exit] device 4.1 still referenced
libusb: warning [libusb_exit] device 2.3 still referenced
libusb: warning [libusb_exit] device 2.15 still referenced
libusb: warning [libusb_exit] device 2.17 still referenced
libusb: warning [libusb_exit] device 2.14 still referenced
libusb: warning [libusb_exit] device 2.13 still referenced
libusb: warning [libusb_exit] device 2.12 still referenced
libusb: warning [libusb_exit] device 2.2 still referenced
libusb: warning [libusb_exit] device 2.9 still referenced
libusb: warning [libusb_exit] device 2.7 still referenced
libusb: warning [libusb_exit] device 2.5 still referenced
libusb: warning [libusb_exit] device 2.1 still referenced
libusb: warning [libusb_exit] device 3.1 still referenced
libusb: warning [libusb_exit] device 1.1 still referenced
[INFO @ host/libraries/libbladeRF/src/helpers/version.c:103] FPGA version (v0.15.0) is newer than entries in libbladeRF's compatibility table. Please update libbladeRF if problems arise.
(RF) Gain: 60.0 (Requested 10)
WARNING: Gain if not supported by source!
Supported gains: ['system', 'full']
WARNING: Gain bb not supported by source!
Supported gains: ['system', 'full']
Bandwidth: 10000000.0 (Requested 10000000)
Warning: Setting antenna to RX1
threshold=18.000000, d_threshold=0.071648 (36.683563/512)
d_max_bursts=240
1679211905 | i:   0/s | i_avg:   0/s | q_max:    0 | i_ok:   0% | o:    0/s | ok:   0% | ok:   0/s | ok_avg:   0% | ok:          0 | ok_avg:   0/s | d: 0

I mean, is this related to the libusb warnings I'm getting? or perhaps the libbladeRF problems? I thought this, but since it does the same thing with the hackrf I doubt it. Here is from the hackrf:
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.10.1.1
built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Using HackRF One with firmware v1.6.0
(RF) Gain: 0.0 (Requested 10)
IF Gain: 40.0 (Requested 40)
BB Gain: 20.0 (Requested 20)
Bandwidth: 10000000.0 (Requested 10000000)
Warning: Setting antenna to TX/RX
1679212686 | i:   0/s | i_avg:   0/s | q_max:    0 | i_ok:   0% | o:    0/s | ok:   0% | ok:   0/s | ok_avg:   0% | ok:          0 | ok_avg:   0/s | d: 0
1679212687 | i: 199/s | i_avg: 188/s | q_max:   96 | i_ok:   0% | o:  191/s | ok:   0% | ok:   0/s | ok_avg:   0% | ok:          0 | ok_avg:   0/s | d: 0
1679212688 | i: 974/s | i_avg: 570/s | q_max:  116 | i_ok:   0% | o: 1050/s | ok:   0% | ok:   6/s | ok_avg:   0% | ok:          7 | ok_avg:   3/s | d: 0
1679212689 | i: 1270/s | i_avg: 799/s | q_max:  116 | i_ok:   0% | o: 1460/s | ok:   0% | ok:   9/s | ok_avg:   0% | ok:         17 | ok_avg:   5/s | d: 0
1679212690 | i: 1044/s | i_avg: 859/s | q_max:  119 | i_ok:   1% | o: 1217/s | ok:   1% | ok:  10/s | ok_avg:   0% | ok:         28 | ok_avg:   6/s | d: 0
1679212691 | i: 1303/s | i_avg: 947/s | q_max:  143 | i_ok:   1% | o: 1339/s | ok:   1% | ok:  17/s | ok_avg:   0% | ok:         46 | ok_avg:   9/s | d: 0
^C1679212692 | i: 1008/s | i_avg: 957/s | q_max:  122 | i_ok:   0% | o: 1270/s | ok:   0% | ok:   7/s | ok_avg:   0% | ok:         54 | ok_avg:   8/s | d: 0
WARNING: your SDR seems to be losing samples. ~250k samples lost (2%)

Thanks in advance for any help you might be able to give me, and once again thank you for all your hard work on this project.

--James Mackay (jamesmac10)

Sec42, 2023-03-19:
Not sure how you say it's way off? Do you mean the "1679212691" at the beginning of the line? That is a timestamp, not a frequency. (See https://github.com/muccc/gr-iridium#sdr--live-mode )

jamesmac10, 2023-03-19:
Oh wow, how did I miss that? I just counted the digits and assumed it was
frequency. Oof! I apologize for wasting your time. But thank you for your
clarification.

closed 2023-03-19 by Sec42