muccc/gr-iridium #174: pybombs installation with gr 3.10

dg9bja, 2023-03-25:
Is it possible that gr-iridum does not work with a gr 3.10 installation? I tried that procedure with pybombs. Last year beginning of november it worked for me. 

I made screenshots of cmake -B build of both systems. After Found volk I see different lines. I am not an expert in this topic. 

Anyone who could help?

[image]

Sec42, 2023-03-26:
Not completely sure what you are doing. You write about cmake -B, but using pybombs you usually do not call cmake yourself.

Can you post the exact steps you did (including the gnuradio installation)?

Alternatively, did you try the steps from https://github.com/Sec42/sec-gr ?

dg9bja, 2023-03-26:
First I tried steps from Sec42 with pybombs installation. I did not see any error there. But after trying to run iridium-extractor I get an error with module iridium missing. 
So I tried to build gr-iridium manually to see what happens.
Did compare GR/lib/python3.9/site-packages/ of my running system and the new one. Directory iridium completely missing and in gnuradio only a subdirectory. 
Python and GRC components are disabled (output of cmake at my new system).

Sec42, 2023-04-06:
I just tried a new install yesterday with pybombs according to the instructions in the sec-gr repo on my ubuntu 22.04 LTS with python3.10 installation & it worked normally.

As I have no further information from you & I can't recreate it, I don't think I'll be able to help you.

dg9bja, 2023-04-08:
Okay thanks for testing. Maybe I need to try Ubuntu then instead of my actual Debian system.

alphapats, 2024-02-13:
Error when running gr-iridium (gnuradio3.10 and ubuntu22.4)

iridium-extractor -D 4 gr-iridium/examples/hackrf.conf
terminate reached from thread id: 7eff23651000Got std::runtime_error
rpcmanager: Aggregator not in use, and a rpc booter is already registered
0x7eff229d530f: (gr::tagged_stream_block::check_topology(int, int)+0x24bf)
0x7eff226ae20c: (std::rethrow_exception(std::__exception_ptr::exception_ptr)+0x7c)
0x7eff226ae277: (std::terminate()+0x17)
0x7eff226ae4d8: (__cxa_throw+0x48)
0x7efe958e4ce4: (gr::local_sighandler::throw_signal(int)+0x9d9e)
0x7efe958f0866: (gr::local_sighandler::throw_signal(int)+0x15920)
0x7eff237a947e: (__nptl_change_stack_perm+0x12ce)
0x7eff237a9568: (__nptl_change_stack_perm+0x13b8)
0x7eff23574af5: (_dl_catch_exception+0xe5)
0x7eff237b0ff6: (_dl_rtld_di_serinfo+0x3096)
0x7eff23574a98: (_dl_catch_exception+0x88)
0x7eff237b134e: (_dl_find_dso_for_object+0x17e)
0x7eff2349063c: (dlmopen+0xfc)
0x7eff23574a98: (_dl_catch_exception+0x88)
0x7eff23574b63: (_dl_catch_error+0x33)
0x7eff2349012e: (dlerror+0x2ae)
0x7eff234906c8: (dlopen+0x48)
0x564be8ebb4fb: (_Py_fopen_obj+0x18b)
0x564be8eb9a27: (PyUnicode_Tailmatch+0x387)
0x564be8db4c59: (PyCell_New+0xc9)
0x564be8d9f5d7: (_PyEval_EvalFrameDefault+0x2a27)
0x564be8db49fc: (_PyFunction_Vectorcall+0x7c)
0x564be8da2cfa: (_PyEval_EvalFrameDefault+0x614a)
0x564be8db49fc: (_PyFunction_Vectorcall+0x7c)
0x564be8d9d45c: (_PyEval_EvalFrameDefault+0x8ac)
0x564be8db49fc: (_PyFunction_Vectorcall+0x7c)
0x564be8d9d26d: (_PyEval_EvalFrameDefault+0x6bd)
0x564be8db49fc: (_PyFunction_Vectorcall+0x7c)
0x564be8d9d26d: (_PyEval_EvalFrameDefault+0x6bd)
0x564be8db49fc: (_PyFunction_Vectorcall+0x7c)
0x564be8d9d26d: (_PyEval_EvalFrameDefault+0x6bd)
0x564be8db49fc: (_PyFunction_Vectorcall+0x7c)
0x564be8db3dc9: (PyObject_CallFunctionObjArgs+0x1e9)
0x564be8e947cf: (_PyObject_CallMethodIdObjArgs+0xff)
0x564be8dc87e3: (PyImport_ImportModuleLevelObject+0x4b3)
0x564be8d9fea3: (_PyEval_EvalFrameDefault+0x32f3)
0x564be8d999c6: (_PyArg_ParseTuple_SizeT+0x696)
0x564be8e8f256: (PyEval_EvalCode+0x86)
0x564be8e94e2d: (PyFrozenSet_New+0x17d)
0x564be8db4c59: (PyCell_New+0xc9)
0x564be8d9f5d7: (_PyEval_EvalFrameDefault+0x2a27)
0x564be8db49fc: (_PyFunction_Vectorcall+0x7c)
0x564be8da2cfa: (_PyEval_EvalFrameDefault+0x614a)
0x564be8db49fc: (_PyFunction_Vectorcall+0x7c)
0x564be8d9d45c: (_PyEval_EvalFrameDefault+0x8ac)
0x564be8db49fc: (_PyFunction_Vectorcall+0x7c)
0x564be8d9d26d: (_PyEval_EvalFrameDefault+0x6bd)
0x564be8db49fc: (_PyFunction_Vectorcall+0x7c)
0x564be8d9d26d: (_PyEval_EvalFrameDefault+0x6bd)
0x564be8db49fc: (_PyFunction_Vectorcall+0x7c)
0x564be8db3dc9: (PyObject_CallFunctionObjArgs+0x1e9)
0x564be8e947cf: (_PyObject_CallMethodIdObjArgs+0xff)
0x564be8dc87e3: (PyImport_ImportModuleLevelObject+0x4b3)
0x564be8da5155: (_PyEval_EvalFrameDefault+0x85a5)
0x564be8db49fc: (_PyFunction_Vectorcall+0x7c)
0x564be8da9cbd: (_PyObject_FastCallDictTstate+0x16d)
0x564be8dbea64: (_PyStack_AsDict+0x2f4)
0x564be8daae2b: (_PyObject_MakeTpCall+0x60b)
0x564be8dc342b: (PyObject_Call+0xbb)
0x564be8d9f5d7: (_PyEval_EvalFrameDefault+0x2a27)
0x564be8d999c6: (_PyArg_ParseTuple_SizeT+0x696)
0x564be8e8f256: (PyEval_EvalCode+0x86)
0x564be8eba108: (PyUnicode_Tailmatch+0xa68)
0x564be8eb39cb: (PyInit__collections+0x58b)
0x564be8eb9e55: (PyUnicode_Tailmatch+0x7b5)
0x564be8eb9338: (_PyRun_SimpleFileObject+0x1a8)
0x564be8eb8f83: (_PyRun_AnyFileObject+0x43)
0x564be8eaba5e: (Py_RunMain+0x2be)
0x564be8e8202d: (Py_BytesMain+0x2d)
0x7eff23429d90: (__libc_init_first+0x90)
0x7eff23429e40: (__libc_start_main+0x80)
0x564be8e81f25: (_start+0x25)
Aborted (core dumped)

schneider42, 2024-03-14:
@dg9bja is this still an issue for you?

dg9bja, 2024-03-17:
I give it up to compile by myself. I used DragonOS to use it.

closed 2024-03-17 by dg9bja