muccc/gr-iridium #175: A liitle question

xyj8, 2023-03-27:
Hi!

I have learned the gr-iridium code for a long time, but I have a few small questions that I would like to consult. Thank you very much for reading my questions.

A burst signal will be caught when the program runs "fft_burst_tagger_impl.cc". May I ask if you are targeting the Ring Alert channel in the Iridium simplex downlink channel? Or the entire iridium signal. May I take the liberty to ask if you have a specific capture process to show?

Then I would like to ask, in the “fft_burst_tagger_imp. cc" file, set an array named "d_burst_mask_f", I have not yet understood the specific purpose of it. Could you explain what it does?

And finally, for an Iridium signal like the one I showed you, is its Ring Alert periodically sent to the ground?

Thank you very much for reading!
[image]

xyj8, 2023-04-17:
> 你好！
> 
> 我已经学习了很长时间的gr-铱代码，但我有几个小问题想咨询一下。非常感谢您阅读我的问题。
> 
> 当程序运行“fft_burst_tagger_impl.cc”时，将捕获突发信号。请问您是否针对铱星单工下行链路通道中的振铃警报通道？或者整个铱星信号。我可以冒昧地问一下您是否有特定的捕获过程要展示？
> 
> 然后我想问一下，在“fft_burst_tagger_imp.cc”文件中，设置一个名为“d_burst_mask_f”的数组，我还不明白它的具体用途。你能解释一下它的作用吗？
> 
> 最后，对于像我给你们展示的铱星信号，它的环形警报会定期发送到地面吗？
> 
> 非常感谢您的阅读！ [image]

I'm glad I've figured out most of the logic. But I would like to ask you about the signal threshold. Why use 512ms of data as an estimate of background noise?

xyj8, 2023-04-17:
I'm glad I've figured out most of the logic. But I would like to ask you about the signal threshold. Why use 512ms of data as an estimate of background noise?

schneider42, 2024-03-14:
The 512 ms really just are a random number which felt reasonable at the time. There is no specific reasoning behind it. Any other number might just work as well.

closed 2024-04-12 by schneider42