muccc/gr-iridium #176: GR-Iridium with RSPdx (SDRPlay) not functioning

angryBird87, 2023-04-06:
Hello, I am currently attempting to get MUCCC gr-iridium running with my RSPdx, but have run into several issues.

Firstly, let me preface, that my hardware is working perfectly fine. I have installed the SDRPlay 3 API, SoapySDR and SoapySDRPlay3 on DragonOS. The RSPdx works well with SDR++ on this device and also has worked on other devices in the past (not with gr-iridium though).

Attached you find the screenshots of the output the command `iridium-extractor -D 4 sdrplay.conf` has produced and the sdrplay.conf file I have written for gr-iridium. As you can see, iridium-extractor seems to ignore my configuration entirely, by for instance always setting the bandwidth to 0.0 and throwing other errors. I would appreciate it, if anyone could help me getting gr-iridium running. Thanks in advance!

[image]
The output produced by the command

[image]
My custom SDRPlay conf file

Sec42, 2023-04-06:
You can try to verify if the osmosdr-source works for you, for example by running osmocom_fft and see if it works (including changing some settings). If osmocom_fft works, iridium-extractor should also work.

If you can't get it to work, you can always try it with a soapysdr source. You would need to install the proper soapy module for your sdr  - which seems to be this one: https://github.com/pothosware/SoapySDRPlay3

Of course you'll also need to make a new config file based on one of the "XXX-soapy.conf" templates.

angryBird87, 2023-04-08:
Using SoapySDR did the trick. Thanks for your help!

closed 2023-04-08 by angryBird87

Botox200, 2024-11-01:
@angryBird87
Tried everything without success. Could you please post the .conf file for the RSP1A?

thx