muccc/gr-iridium #178: burst detection and duration

yangyaxun, 2023-05-28:
Hi,

I hope this message finds you well. I have a question regarding the identification of burst duration, including preamble and unique word duration, in an Iridium signal, as well as handling the scenario of multiple bursts occurring simultaneously on the same frequency channel. Specifically, I am interested in addressing situations where there might be two or three ring alert bursts simultaneously, albeit with slightly different center frequencies due to the Doppler effect. I would appreciate your insights on how to recognize and differentiate these bursts in either the frequency or time domain.

Thank you in advance for your assistance. I look forward to your response.

Best regards

Sec42, 2023-05-29:
Hi,

Not sure what your question exactly is. The case that you are describing is what the existing code in gr-iridium is written for.

There is not much documentation on the details besides the code itself. The burst detection and tagging happens here: https://github.com/muccc/gr-iridium/blob/master/lib/fft_burst_tagger_impl.cc#L524

If you're interested in more of the design of gr-iridium, you can watch this talk by schneider form 2018: https://www.youtube.com/watch?v=6ngYp8W-AX0

yangyaxun, 2023-05-30:
Thank you for your prompt response and apologies for the ambiguous question.

I'm curious to know how you would handle the situation when there are multiple satellites transmitting multiple channels, such as ring alert and primary message, and you need to channelize them. It appears that there are no signal characteristics in the time domain or frequency domain that can differentiate the signals from specific satellites before decoding them (like the CA code of GPS).

Currently, I'm working on signal processing of Iridium using USRP X300 and GNU Radio for receiving the Iridium signal (only simplex channel), and I'm processing it in MATLAB. However, I encounter difficulties when channelizing the signal in the presence of two or three satellites transmitting simultaneously. In the attached figure, the 0 frequency corresponds to 1626.25 MHz in terms of baseband. There are two peaks in the ring alert frequency range and the primary message frequency range, indicating that two Iridium satellites are simultaneously transmitting ring alert and primary message signals. How would you separate and distribute these signals to different channels?

I appreciate your assistance in advance and eagerly await your response.

Best regards

[image]

Sec42, 2023-06-04:
I suggest you watch the above linked talk by schneider. It is the best documentation we have on how gr-iridium detects and handles the signals.

yangyaxun, 2023-06-04:
I have been trying to understand the code and have watched the linked talk by schneider. Thanks for responding !

schneider42, 2024-03-14:
> How would you separate and distribute these signals to different channels?

Well that is very basic signal processing. Either you first shift and then apply a low-pass filter or you apply a band-pass filter and then shift.

Of course there are also options to directly channelize it into multiple channels using a channelizer. That is what gr-iridium does to save some compute power. We use a custom implementation of an FFT based channelizer. I don't claim correctness of that one. It works good enough for gr-iridium though :)

schneider42, 2024-03-14:
Obviously https://media.ccc.de/v/gpn18-15-channelizing-with-gnuradio is a good source on how the FFT channelizer works in general.

schneider42, 2024-04-12:
Let us know if you have further questions.

closed 2024-04-12 by schneider42