muccc/gr-iridium #181: VOLK version mismatch and compile Error

Infiltra, 2023-08-24:
So when compiling the maint3.8 branch there is a compile error due to an improper mismatch of VOLK versions. Because of this compilation is failing. But the file <volk/volk_alloc.hh> if put in include/iridium path it works. A git diff would look like this

```
diff --git a/lib/burst_downmix_impl.h b/lib/burst_downmix_impl.h
index 83619ab..9d8112f 100644
--- a/lib/burst_downmix_impl.h
+++ b/lib/burst_downmix_impl.h
@@ -26,7 +26,8 @@
 
 #include <iridium/burst_downmix.h>
 
-#include <volk/volk_alloc.hh>
+#include <iridium/volk_alloc.hh>
+// #include <volk/volk_malloc.h>   
 
 namespace gr {
   namespace iridium {
```
The fix is a temp fix, the ideal fix would be to check volk major and minor versions and include the files accordingly.

schneider42, 2024-03-14:
Before diving into this: Do you still use GNURadio 3.8? Which environment is still based on that?

schneider42, 2024-04-12:
Going to close this. Please let us know if this is still an issue.

closed 2024-04-12 by schneider42