muccc/gr-iridium #182: UNIX timestamp can not output

xyj8, 2023-10-10:
Hello, when I was using the IRIDIUM Toolkit Toolkit to demodulation data from the Iridium satellite network, my terminal reminded me" Warning: no timestamp found in filename". Then I went to the frame form website to query the specific structure and found that I could not output UNIX timestamp, how to solve this problem?
[image]

Sec42, 2023-10-10:
This message is generated by iridium-parser if the format in the second field of the .bits file does not match any known format.

see https://github.com/muccc/gr-iridium#bits-output and the corresponding footnote

It is likely an issue with how you ran iridium-extractor. If you can share the first RAW: lines of the output.bits file, I can probably say more.

xyj8, 2023-10-10:
> This message is generated by iridium-parser if the format in the second field of the .bits file does not match any known format.
>
> see https://github.com/muccc/gr-iridium#bits-output and the [...]

The first diagram shows the intermediate frequency, sampling rate, file path, and output file name of the iridium-extrator I used. Then the second picture shows the content part of the specific output.bit file. Please tell me where the use is wrong. I would also like to ask whether this output UNIX timestamp is the absolute time when the receiver starts collecting data? Thank you for your sharing, I appreciate it.

[image]
[image]

Sec42, 2023-10-10:
You were not technically doing anything wrong. It's just that when you run iridium-extractor on an existing recording it doesn't know when that recording was made. So everything is working as intended.

The timestamp would be automatically set by iridium-extractor when you use one of the "online" modes (as described [in the README](https://github.com/muccc/gr-iridium#online-with-an-sdr)).

iridium-parser will still parse these bits just fine, no info should be lost - you will just lack the date/time info for each line.

If you want to manually specify the time at which this recording was made, you could use the `--file-info` commandline option of iridium-extractor and specify "`i-<timestamp>-t1`" where you replace `<timestamp>` with the unix time_t of the start of your recording.

xyj8, 2023-10-10:
> You were not technically doing anything wrong. It's just that when you run iridium-extractor on an existing recording it doesn't know when that recording was made. So everything is working as [...]

Thank you very much! When I first reply to you , I read the gr-iridium document carefully again.I also noticed this oversight of mine by using "offline" mode to run iridium-extrator.So I would like to ask if I want to run the program in "online" mode, do I need to connect my USRP x300 and run Iridim-Extractor while collecting the raw signal? Is the" `iridium-extractor -D 4 examples/hackrf.conf > output.bits` "used to run?

Sec42, 2023-10-10:
I mean, hackrf.conf will not work if you don't use a hackrf. I suggest checking the documentation about the osmosdr-souce or soapy-source in the README and create an appropriate file for your SDR setup.

closed 2024-03-14 by Sec42