muccc/gr-iridium #190: Remove scipy from iridium_extractor_flowgraph.py?

jmfriedt, 2024-03-19:
Is the ``import scipy`` in
https://github.com/muccc/gr-iridium/blob/master/python/iridium_extractor_flowgraph.py
needed? I have removed it and the flowgraph seeems to run happily. Removing scipy removes a huge dependency for running gr-iridium on embedded boards (e.g. Raspberry Pi 4) running a Buildroot generated operating system.

Sec42, 2024-03-30:
I seem to remember that it was needed for some debugging output/feature. Checking the git history I can't recall any details though. Maybe @schneider42 remembers something?

schneider42, 2024-04-12:
This must be a leftover from the pure Python implementation. I'm going to remove it.

schneider42, 2024-04-12:
Pushed the change directly to master. Thanks.

closed 2024-04-12 by schneider42